gbionics/whole-body-estimators #17: [wholeBodyDynamics] add the option to change the type of `iDynTree::UnknownWrenchContact`

HosameldinMohamed, 2020-02-01:
It would be nice if we can assume that only external forces acting on the torso are the thrust forces, for which we know the application point and the force direction, So we can estimate the intensities of more than one wrench per sub-model.

According to the estimation theory behind the algorithm (see the paper [Contact Force Estimations Using Tactile Sensors and Force Torque Sensors](https://www.researchgate.net/profile/Andrea_Del_Prete/publication/236152161_Contact_Force_Estimations_Using_Tactile_Sensors_and_Force_Torque_Sensors/links/00b495166e74a5369d000000.pdf). If the applied external wrenches are assumed to `PURE_FORCE`, the estimator can estimate up to 3 forces per sub-model, and if the applied forces are assumed to be `PURE_FORCE_WITH_KNOWN_DIRECTION`, the estimator can estimate up to 6 forces per sub-model.

The idea is to change the `wholeBodyDynamics` device (Without forcing the software that depends on it to be changed) to have the option to specify the unknown external wrenches from the configuration files.

## Proposed modifications

Here are the proposed modifications to the `wholeBodyDynamics` device:

- The idea is to add some parameters in the configuration file `wholebodydynamics-external.xml` ([example](https://github.com/robotology/codyco-modules/blob/de4546e3f0e4bbf89fc474ccd578febf7836d517/src/devices/wholeBodyDynamics/app/wholebodydynamics-icub-eth-six-fts.xml#L9)) right below the [`defaultContactFrames` parameter](https://github.com/robotology/codyco-modules/blob/de4546e3f0e4bbf89fc474ccd578febf7836d517/src/devices/wholeBodyDynamics/app/wholebodydynamics-icub-eth-six-fts.xml#L9).

The additional parameters could be:

- `overrideContactFrames`:
Similar to `defaultContactFrames` contains vector of strings (name of frames). If this parameter exists, the device will ignore the parameter `defaultContactFrames`.

- `contactWrenchType` could be one of the types:
full = FULL_WRENCH
pure = PURE_FORCE
pureKnown = PURE_FORCE_WITH_KNOWN_DIRECTION

- `contactWrenchDirection`:
Which is a vector of 3 real numbers to specify the direction of the contact wrench w.r.t the origin of the corresponding  `overrideContactFrames` frame. They must be filled if the corresponding `contactWrenchType` is `PURE_FORCE`, and be left empty if the type is `FULL_WRENCH` or `PURE_FORCE_WITH_KNOWN_DIRECTION`.

- `contactWrenchPosition`:
Which is a vector of 3 real numbers to specify the position of the contact wrench w.r.t the origin of the corresponding  `overrideContactFrames` frame. This parameter is **optional**. If left empty, the external force will be considered applied to the origin of the link as usual.

## Logic

- If the configuration file doesn't contain the parameter `overrideContactFrames`. It continues, as usual, considering `defaultContactFrames` and assuming `FULL_WRENCH` is applied.

- If the configuration file contains the parameter `overrideContactFrames`. It ignores frames in the parameter `defaultContactFrames` and considers `overrideContactFrames`. It then requires the presence of the next 2 parameters: `contactWrenchType`, and `contactWrenchDirection`

- The device shall check the consistency of data contained in the 3 paramteres: `overrideContactFrames`, `contactWrenchType` and `contactWrenchDirection`.

## Consistency checks

- For each frame in the paramater `overrideContactFrames`, there is one string value in the paramater `contactWrenchType` (one of the 3 types mentioned above), and 3 real numbers in the paramater `contactWrenchDirection`.

- The frames from a single submodel should be no more than 1 if the type is `FULL_WRENCH`, no more than 3 if the type is `PURE_FORCE`, and no more than 6 if the type is `PURE_FORCE_WITH_KNOWN_DIRECTION`.

- Proper storage of the values read from the configuration file.

- Proper filling the object `UnknownWrenchContact`.

HosameldinMohamed, 2020-03-13:
I'd like to open the PR.

- Is it better to merge it to `master` or `devel`?

- Is it better to create a branch based on [include_FT_Temperature_Offset](https://github.com/robotology/whole-body-estimators/tree/include_FT_Temperature_Offset)?

prashanthr05, 2020-03-13:
I think you can open a PR against `devel` and maybe we can rebase [include_FT_Temperature_Offset](https://github.com/robotology/whole-body-estimators/tree/include_FT_Temperature_Offset) over the latest `devel`?

What do you think @traversaro @fjandrad ?

traversaro, 2020-03-13:
I agree.

fjandrad, 2020-03-16:
So to understand the workflow, we will open the PR for these changes to devel, merge that PR and then I rebase the PR for the temperature offset on top of it right?

prashanthr05, 2020-03-16:
Whichever PR is merged first. The other PR will be rebased on top of the merged devel and proceeded with :D

fjandrad, 2020-03-23:
Probably that will be after testing with the robot....

HosameldinMohamed, 2020-04-11:
I found that in the **publishing part** of `wholeBodyDynamics` device:
https://github.com/robotology/whole-body-estimators/blob/22302eef59ad2f0b900a662490850206a6aea909/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L420
 I didn't take it into account that the device publishes the `netExternalWrenchesExertedByTheEnviroment` by summing all the exerted external wrenches.
https://github.com/robotology/whole-body-estimators/blob/22302eef59ad2f0b900a662490850206a6aea909/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1822

This was OK before, since we assume only a single wrench for each sub-model and hence for each link. But for the proposed changes there could be more than one external wrench per link.
So instead of https://github.com/robotology/whole-body-estimators/blob/22302eef59ad2f0b900a662490850206a6aea909/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1831
We need to publish
```cpp
iDynTree::Wrench & link_f = estimateExternalContactWrenches.contactWrench(link,j).contactWrench();
```
`j` is the index of the contact wrench in the link `link`. 

- The publishing yarp ports shall have sizes more than 6 in case of multiple contacts per link.

traversaro, 2020-04-11:
> 
> 
> I found that in the **publishing part** of `wholeBodyDynamics` device:
> 
> https://github.com/robotology/whole-body-estimators/blob/22302eef59ad2f0b900a662490850206a6aea909/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L420
> 
> I didn't take it into account that the device publishes the `netExternalWrenchesExertedByTheEnviroment` by summing all the exerted external wrenches.
> 
> https://github.com/robotology/whole-body-estimators/blob/22302eef59ad2f0b900a662490850206a6aea909/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1822
> 
> This was OK before, since we assume only a single wrench for each sub-model and hence for each link. But for the proposed changes there could be more than one external wrench per link.
> So instead of
> 
> https://github.com/robotology/whole-body-estimators/blob/22302eef59ad2f0b900a662490850206a6aea909/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1831
> 
> We need to publish
> 
> ```c++
> iDynTree::Wrench & link_f = estimateExternalContactWrenches.contactWrench(link,j).contactWrench();
> ```
> 
> `j` is the index of the contact wrench in the link `link`.
> 
>     * The publishing yarp ports shall have sizes more than 6 in case of multiple contacts per link.

Hi @HosameldinMohamed, these is  one of the several publishing mechanisms of wholeBodyDynamics. For links with multiple contacts, detailed information of each contact is currently published on the `/wholeBodyDynamics/contacts:o` port, see https://github.com/robotology/whole-body-estimators/blob/22302eef59ad2f0b900a662490850206a6aea909/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L385 . If you want to publish detailed information of each contacts in a different way (for example as a  vector to which it is easier to read from Simulink)  that it is ok, but please make sure that it is still possible to get the old behaviour of reading the `netExternalWrenchesExertedByTheEnviroment` info, for example with an  option, thanks!

HosameldinMohamed, 2020-04-11:
> Hi @HosameldinMohamed, these is one of the several publishing mechanisms of wholeBodyDynamics. For links with multiple contacts, detailed information of each contact is currently published on the /wholeBodyDynamics/contacts:o port

Thanks @traversaro  for pointing this out. But as you said, their format may not be suitable for using it in Simulink for example.

> If you want to publish detailed information of each contacts in a different way (for example as a vector to which it is easier to read from Simulink) that it is ok, but please make sure that it is [...]

My idea is to publish
```cpp
iDynTree::Wrench & link_f = estimateExternalContactWrenches.contactWrench(link,j).contactWrench();
```
in the same ports in the group [WBD_OUTPUT_EXTERNAL_WRENCH_PORTS](https://github.com/robotology/whole-body-estimators/blob/22302eef59ad2f0b900a662490850206a6aea909/devices/wholeBodyDynamics/app/wholebodydynamics-icub-eth-six-fts.xml#L43).

Because in the case of using `defaultContactFrames` or `overrideContactFrames` with `contactWrenchType`=`full`. The published contact wrench is equal to the `net external wrench`.
And with using `overrideContactFrames` with `contactWrenchType`=`pure` or `pureKnown`. The same port will increase the size.

With this I think the old behavior will not change, but the person that uses the new functionality should know how the data will be published.

traversaro, 2020-04-11:
> Because in the case of using `defaultContactFrames` or `overrideContactFrames` with `contactWrenchType`=`full`. The published contact wrench is equal to the `net external wrench`.
> And with using [...]

That is only true if no data is read from the `/wholeBodyDynamics/skin_contacts:i` port. If some data is read from `/wholeBodyDynamics/skin_contacts:i`, the size of the  ports in the group WBD_OUTPUT_EXTERNAL_WRENCH_PORTS would be changing over time, and that would be difficult to deal with. For this  reason, I suggest to implement this functionality by guarding it under a specific option.

HosameldinMohamed, 2020-05-20:
PR #48 is merged.

Closing.

closed 2020-05-20 by HosameldinMohamed