gbionics/whole-body-estimators #32: Specify used F/T sensors in wholebodydynamics with an explicit parameter

traversaro, 2020-02-01:
TODO 

cc @gabrielenava @ahmadgazar remember me to update this issue

fjandrad, 2020-04-17:
For developing the temperature compensation the `multipleAnalogSensorsNames` variable was inserted, it allows to signal which are the sensors that are a MAS. We might include a similar variable for selecting which sensors to use. Things to keep in mind are:
- To keep a similar logic as the other variables we will add the sensors to be considered.
- The new variable should have priority over `multipleAnalogSensorsNames`
- Handle the case where `multipleAnalogSensorsNames` could be a valid name but it was removed by the new variable. Should send a warning or fail if a name is inserted that is not there?

traversaro, 2020-04-17:
> Should send a warning or fail if a name is inserted that is not there?

Fail. [Fail early, fail fast.](https://en.wikipedia.org/wiki/Fail-fast)

fjandrad, 2020-05-28:
@prashanthr05 this is the issue about enabling/disabling the ft sensors

traversaro, 2020-11-24:
@prashanthr05 this is probably related to the problem of excluding the FT sensors on the hip that we were discussing yesterday.

prashanthr05, 2020-11-24:
> I think this is meant to allow to enable/disable and FT sensor from the perspective of the estimation without the need to modify the urdf model.

Ideally, this would be a nice feature to have. 
But commenting out the sensors in the URDF seems to be a workaround solution for quick hacks.

S-Dafarra, 2024-03-01:
Let me revive this issue, as it happens often, also during demos, that we need to avoid using an F/T from the robot. Editing the URDF is usually something not so convenient

fjandrad, 2024-03-01:
Wow its been a while. This takes me back haha. I think I was very close to having this ready or that might just me editing my past memories. What you want to do might be related to this function https://github.com/robotology/whole-body-estimators/blob/5fb9aeeb1cc9282189eee11754c78f685814ed9f/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L767 . I vaguely remember activating one or the other depending if a sensor had the temperature component or not.

fjandrad, 2024-03-01:
It could also be that you can modify the sensorlist sent to the estimator in here https://github.com/robotology/idyntree/blob/759fd719b3ad7f6c63f565d99250dd7174177161/src/estimation/src/ExtWrenchesAndJointTorquesEstimator.cpp#L76 . At the moment I do not remember where the sensorlist comes from, I guess there is a configuration file somewhere.

traversaro, 2024-03-01:
Thanks @fjandrad !

> Let me revive this issue, as it happens often, also during demos, that we need to avoid using an F/T from the robot. Editing the URDF is usually something not so convenient

As discussed in person, probably we another thing we could implement that would be convenient is to actually get the FT sensors to use from YARP instead of from the URDF. In general you would not like to do that (as if there is some kind of error on the YARP side, you only use a subset of the sensors without realized that), but in case some FT sensors are missing you want to do that.

closed 2024-07-12 by traversaro