gbionics/whole-body-estimators #40: Unable to run the RVIZ visualization

GiulioRomualdi, 2020-02-16:
I'm trying to run the RVIZ visualization located in `whole-body-estimators/devices/baseEstimatorV1/ros` 
The command I'm using is `roslaunch fbeViz.launch `
An I'm getting the following error.
```
roslaunch fbeViz.launch 
... logging to /home/gromualdi/.ros/log/8762f73a-50ca-11ea-ab11-9cb6d0f6860f/roslaunch-iiticublap109-23145.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: Invalid <arg> tag: iCub
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/opt/ros/melodic/share. 

Arg xml is <arg default="$(find iCub)/robots/$(env YARP_ROBOT_NAME)/model.urdf" name="model"/>
The traceback for the exception was written to the log file
```

I'm pretty sure I'm making some mistakes (I'm not a ROS user :cry: )

Can you help me? cc @prashanthr05

prashanthr05, 2020-02-16:
@GiulioRomualdi  please excuse for the lack of documentation on the visualization using RViz. I will update it. 

However, I feel rusty with the code as well. I need to look into it and try to fix it if its missing some crucial parts.

I suppose, the solution depends on where the repository `icub-models` is installed. Could you make sure that `ROS_PACKAGE_PATH` points to where `icub-models` is installed? 

For more details, refer to https://github.com/robotology/icub-models

traversaro, 2020-02-16:
Probably we need to add `ROS_PACKAGE_PATH` to the env variables exported in the superbuild's env variables scripts.

Ekanshh, 2020-02-18:
I think the issue is with the command you're writing, try using this command: 
` roslaunch <package name> <launch file>`

Yeshasvitvs, 2020-02-19:
@GiulioRomualdi one quick check is to use [roscd](http://wiki.ros.org/rosbash#roscd) to check if ros can correctly discover your `ros-package`, in which you have the launch file. Let me know if you need a hand to run rviz.