gbionics/whole-body-estimators #41: Work on scheduled upgrades to wholebodydynamics device

fjandrad, 2020-03-10:
Following the [transfer of wholebodydynamics device into this repo](https://github.com/robotology/whole-body-estimators/pull/16) and its incoming inclusion in [robotology-superbuild](https://github.com/robotology/robotology-superbuild/issues/365) we need to add the work scheduled for it in this repo.

The scheduled changes to wholebodydynamics device were:

- [ ] [Add support for compensating temperature in Force/Torque sensors readings and for specify user offline offset ](https://github.com/robotology/codyco-modules/pull/281)
- [ ] Adding support for the required estimation of forces when using the turbines in the torso area (@HosameldinMohamed feel free to add a better description for this)
- [ ] Refactor the device by splitting to improve readability.

HosameldinMohamed, 2020-08-10:
>  Refactor the device by splitting to improve readability.

We can remove this from this Epic and make it into a separate Epic, since the complexity of the task is higher.

prashanthr05, 2020-08-10:
> We can remove this from this Epic and make it into a separate Epic, since the complexity of the task is higher.

An issue already exists, actually. https://github.com/dic-iit/component_wholebody-teleoperation/issues/241. We could convert it into an epic.

prashanthr05, 2020-09-18:
@HosameldinMohamed shall we close this epic?

HosameldinMohamed, 2020-10-07:
@prashanthr05 
It seems that this is remaining.

> - [ ]  Handle CMake deprecation warnings

I'll open a new issue to continue your work in https://github.com/robotology/whole-body-estimators/pull/58#issuecomment-608299759

HosameldinMohamed, 2020-10-07:
> @prashanthr05
> It seems that this is remaining.
> 
>  Handle CMake deprecation warnings
> I'll open a new issue to continue your work in #58 (comment)

I opened #85, however the remaining deprecation warnings are not specifically related to `wholeBodyDynamics`. So I guess we don't have to connect it to this epic and we can close the epic.

prashanthr05, 2020-10-07:
@HosameldinMohamed thanks for opening the issue. Let's close the epic after addressing that issue.
I added that issue to this epic.

prashanthr05, 2020-11-03:
Thanks for all the upgrades @fjandrad @HosameldinMohamed @traversaro 

We can finally close this epic.

closed 2020-11-03 by prashanthr05