gbionics/whole-body-estimators #43: Merge devel into master and make a new release that includes wholebodydynamics and related devices

prashanthr05, 2020-03-11:
We just recently released a minor version of whole-body-estimators that includes the `baseEstimatorV1` device. We can proceed with another minor version release that includes the `wholebodydynamics` and related devices migrated from `codyco-modules`. This will enable an easy inclusion of whole-body-estimators in the [robotology/robotology-superbuild](https://github.com/robotology/robotology-superbuild).

HosameldinMohamed, 2020-03-11:
As per the T2T meeting with @traversaro, we shall do the following steps:

1. Bump the version in [CMakeLists.txt](https://github.com/robotology/whole-body-estimators/blob/devel/CMakeLists.txt) file.
2. Add a description in the [CHANGELOG](https://github.com/robotology/whole-body-estimators/blob/devel/CHANGELOG.md) file.
3. Merge the `devel` branch into the `master` branch (Adds `wholeBodyDynamics` and related devices to `master`).

CC @prashanthr05 @fjandrad

prashanthr05, 2020-03-11:
@traversaro  just a clarification. Should the points 1 and 2 indicated by @HosameldinMohamed be done in `master` during the release and then forward-merged into `devel`? or is it also a safe practice to do the necessary changes in `devel` itself and then merge into `master` for a release?

HosameldinMohamed, 2020-03-11:
> @traversaro just a clarification. Should the points 1 and 2 indicated by @HosameldinMohamed be done in `master` during the release and then forward-merged into `devel`? or is it also a safe practice to do the necessary changes in `devel` itself and then merge into `master` for a release?

 I think in future, the CHANGELOG file can be modified when introducing the changes (when merging the PR from a branch/fork to `devel`).

HosameldinMohamed, 2020-03-11:
> Now I was thinking to update CHANGELOG and CMakeLists in master after merging.

Actually, now that I think about it, if I do a commit in the `master` after the merge I will have to "forward merge" the `master` into `devel`. So maybe it's better to do one commit in `devel` then merge it to master.

prashanthr05, 2020-03-11:
Yes, probably it would be cleaner and avoid too many circular commits.

traversaro, 2020-03-11:
> 
> @traversaro just a clarification. Should the points 1 and 2 indicated by @HosameldinMohamed be done in `master` during the release and then forward-merged into `devel`? or is it also a safe practice to do the necessary changes in `devel` itself and then merge into `master` for a release?

As you prefer, in the past I did in one way or another, and I do not have a strong preference.

HosameldinMohamed, 2020-03-11:
Release [v0.2.0](https://github.com/robotology/whole-body-estimators/releases/tag/v0.2.0) is added.

prashanthr05, 2020-03-13:
@HosameldinMohamed shall we close this issue?

closed 2020-03-13 by HosameldinMohamed