gbionics/whole-body-estimators #46: wholebodydynamics device does not compile if iDynTree CMake option IDYNTREE_USES_ICUB_MAIN is OFF

prashanthr05, 2020-03-13:
For a report on this, please see https://github.com/prashanthr05/whole-body-estimators/runs/506344138?check_suite_focus=true#step:12:97.

This needs to be documented in the Dependencies section of README.md.

traversaro, 2020-03-14:
Yes, that is definitely inconvenient. My plan was to transform some simple conversions utility that we have in iDynTree in header only libraries, so that we can always install them, and then it is left to the downstream projects to find the dependencies. I could not find a place where to track that, so I created a new issue in https://github.com/robotology/idyntree/issues/653 .

prashanthr05, 2020-03-17:
The README was updated. Closing this issue.

closed 2020-03-17 by prashanthr05