gbionics/whole-body-estimators #50: Document the added variables to use the temperature coefficients

fjandrad, 2020-03-18:
The PR https://github.com/robotology/whole-body-estimators/pull/45 adds the options to use temperature coefficients and offline estimated offset. We will detail the usage of this functions in the wiki in this repo.

fjandrad, 2020-04-17:
Documetation has been added in this [wiki](https://github.com/robotology/whole-body-estimators/wiki/How-to-use-the-temperature-coefficients-and-offset-compensation.) 
Feel free to check and signal if something is not clear. @prashanthr05 @traversaro @S-Dafarra

traversaro, 2020-04-17:
I strongly suggest **not** to use the wiki for this kind of  documentation, but rather make sure that the docs are stored in the repo, eventually as a `.md` file containing exactly the same content of the  wiki page, but inside of the repo. The rationale behind that is that so you always know which version of the software the documentation refers to. For example, let's say that in the future I want to add a new parameters, and I add the software in the `devel` branch, and  it gets merged in `master` only after six months. When I should update the docs, if they are stored in the wiki? If I update it before  `master` have changed, people using the `master` branch may be confused. All this problems are avoided if the docs are put inside the repo.

fjandrad, 2020-05-06:
Pending to move the wiki to a docs folder

prashanthr05, 2020-08-24:
> Pending to move the wiki to a docs folder

I have done this in https://github.com/robotology/whole-body-estimators/pull/84

HosameldinMohamed, 2020-10-07:
PR #84 is merged with `devel`

I close this issue.

PR #84 (2020-10-07): Add howto docs about using temperature compensation and pre-estimated offsets in WBD and add a top-level WBD README
- Move documentation from https://github.com/robotology/whole-body-estimators/wiki/How-to-use-the-temperature-coefficients-and-offset-compensation to `docs` folder
- Moved https://github.com/robotology-legacy/codyco-modules/blob/master/doc/force_control_on_icub.md to `docs` folder  
- Moved doxygen documentation of the `wholeBodyDynamcis` device to a folder level README (removed the doxygen documentation to avoid any future divergence of redundant documentations cc @traversaro's  [... cut]
closed 2020-10-07 by HosameldinMohamed