gbionics/whole-body-estimators #51: WholeBodyDynamics device not launched when running launch-wholebodydynamics.xml

gabrielenava, 2020-03-22:
After I wiped out the `robotology-superbuild` and reinstalled everything from zero, the problem I pointed out in https://github.com/robotology/codyco-modules/issues/277 happened again.

To recap:

>The wholebodydynamics device that should be launched by the `launch-wholebodydynamics.xml` file is actually not launched. Namely, I am talking about this [line](https://github.com/robotology/codyco-modules/blob/master/src/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-six-fts-sim.xml#L95)
>
>Here there are no errors: the line is simply skipped. 

This fix allowed me to run the file:

```
    <devices>
    <xi:include href="estimators/wholebodydynamics-external.xml" />
    </devices>
```

gabrielenava, 2020-03-22:
cc @diegoferigo

prashanthr05, 2020-04-02:
Sadly, the changes from this PR https://github.com/robotology/codyco-modules/pull/280 was not carried over during the migration of the device from codyco-modules https://github.com/robotology/whole-body-estimators/pull/16 and I had missed this criticality in the review.

prashanthr05, 2020-04-02:
A pull request has been opened.

prashanthr05, 2020-04-02:
This was fixed in https://github.com/robotology/whole-body-estimators/pull/56. 

@gabrielenava you can close this issue when you have tested the fix on your local machine.

prashanthr05, 2020-04-22:
@gabrielenava friendly ping! If you will not test this, I will close the issue, because I tested it on my local machine.

gabrielenava, 2020-04-22:
>@gabrielenava friendly ping! If you will not test this, I will close the issue, because I tested it on my local machine.

I will test it, my deadline is tomorrow so I can clean up my laptop configuration (it is a bit messy in these days) and I'll be sure the test is consistent.

gabrielenava, 2020-04-25:
Sorry for the delay, the test was easy but my local configuration was really messy. I had to clean my superbuild and reinstall it to be sure my test was consistent.

Now that I have a  clean and up-to-date superbuild, I tested the fix with `iCubGenovaV2_5`, and it works with no issue! Thanks @prashanthr05!

prashanthr05, 2020-04-27:
Thank you @gabrielenava. Closing this issue.

closed 2020-04-27 by prashanthr05