gbionics/whole-body-estimators #55: Update README to add references to wholebodydynamics

traversaro, 2020-03-26:
The README still contains only references `baseEstimatorV1` , and there is no documentation (or link to documentation) of `wholebodydynamics`.

traversaro, 2020-04-17:
We should also make sure that from the README are reachable: 
* The documentation of the parameters supported by `wholebodydynamics` , that are currently documented in a doxygen block of the `WholeBodyDynamicsDevice` class
* The docs added in https://github.com/robotology/whole-body-estimators/issues/50
* I noticed that we left behind in codyco-modules  the document https://github.com/robotology-legacy/codyco-modules/blob/master/doc/force_control_on_icub.md, perhaps it could make sense to bring it back and store it in this repo.

prashanthr05, 2020-08-24:
@HosameldinMohamed if we could update the READMEs and close this issue, we can make a new release of the `whole-body-estimators`.

prashanthr05, 2020-09-28:
This is done as a part of https://github.com/robotology/whole-body-estimators/pull/84.

HosameldinMohamed, 2020-10-07:
PR #84 is merged to `devel`.

I close this issue.

PR #84 (2020-10-07): Add howto docs about using temperature compensation and pre-estimated offsets in WBD and add a top-level WBD README
- Move documentation from https://github.com/robotology/whole-body-estimators/wiki/How-to-use-the-temperature-coefficients-and-offset-compensation to `docs` folder
- Moved https://github.com/robotology-legacy/codyco-modules/blob/master/doc/force_control_on_icub.md to `docs` folder  
- Moved doxygen documentation of the `wholeBodyDynamcis` device to a folder level README (removed the doxygen documentation to avoid any future divergence of redundant documentations cc @traversaro's  [... cut]
closed 2020-10-07 by HosameldinMohamed