gbionics/whole-body-estimators #60: [workflows] CI macOS build failing in CMake configuration step

prashanthr05, 2020-04-03:
The configuration step is failing with the following error.

``` bash
Call Stack (most recent call first):
  devices/virtualAnalogClient/CMakeLists.txt:9223372036854775807 (_yarp_deprecated_variable_warning)
  devices/virtualAnalogClient/CMakeLists.txt:16 (include_directories)

-- Found YARP: /Users/runner/runners/2.168.0/work/whole-body-estimators/whole-body-estimators/install/deps/lib/cmake/YARP (found version "3.3.2")
--  [x] plugin virtualAnalogRemapper (ENABLE_virtualAnalogRemapper)
CMake Deprecation Warning at install/deps/lib/cmake/YARP/YARPConfig.cmake:258 (message):
  The YARP_INCLUDE_DIRS variable is deprecated
Call Stack (most recent call first):
  devices/virtualAnalogRemapper/CMakeLists.txt:9223372036854775807 (_yarp_deprecated_variable_warning)
  devices/virtualAnalogRemapper/CMakeLists.txt:16 (include_directories)

-- Found YARP: /Users/runner/runners/2.168.0/work/whole-body-estimators/whole-body-estimators/install/deps/lib/cmake/YARP (found version "3.3.2")
--  [x] plugin genericSensorClient (ENABLE_genericSensorClient)
CMake Deprecation Warning at install/deps/lib/cmake/YARP/YARPConfig.cmake:258 (message):
  The YARP_INCLUDE_DIRS variable is deprecated
Call Stack (most recent call first):
  devices/genericSensorClient/CMakeLists.txt:9223372036854775807 (_yarp_deprecated_variable_warning)
  devices/genericSensorClient/CMakeLists.txt:16 (include_directories)

-- Configuring incomplete, errors occurred!
See also "/Users/runner/runners/2.168.0/work/whole-body-estimators/whole-body-estimators/build/CMakeFiles/CMakeOutput.log".
##[error]Process completed with exit code 1.
```

traversaro, 2020-04-03:
This is a problem of icub-main that creates CMake configuration files that are technically speaking non-relocatable, i.e. it cannot be used in a location different from the one in it was originally installed. I suggest to open an issue on icub-main on this. In the meanwhile a possible strategy is to add the  runner version (i.e. the `2.168.0`) in the cache name, so when the runner directory changes the cache is automatically re-generated.

traversaro, 2020-04-03:
An easy workaround is provided in https://github.com/robotology/idyntree/pull/667 , basically you need to add `${{ runner.temp }}` to the cache id, so that it gets automatically invalidated when the path name changes. However, this is not the best solution because it invalidates the cache more then necessary, the best long term solution is to solve https://github.com/robotology/icub-main/issues/651 .

prashanthr05, 2020-04-07:
A temporary fix was done in #62. I will close this issue, for now, we can reopen it when necessary.

PR #62 (2020-04-07): [workflows] Add workaround for non-relocatable ICUB cmake config files
Fixes #60 

Thanks to @traversaro for the fix!!
closed 2020-04-07 by prashanthr05