gbionics/whole-body-estimators #67: Add an option to skip the initial calibration and just use zero offset

traversaro, 2020-04-15:
Users of `whole-body-dynamics` tipically execute the `resetOffset` command at the beginning of the simulation, to  set the offset of the FT sensors  in the    `whole-body-dynamics` to zero. This is both boring and error-prone, as any manual step it may be erroneously skipped.  For this reason, it would make sense to have an option in the `.ini` configuration file of `whole-body-dynamcis`  to automatically skip the initial calibration in https://github.com/robotology/whole-body-estimators/blob/v0.2.1/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1181, and instead use the zero offset (as done in fhttps://github.com/robotology/whole-body-estimators/blob/v0.2.1/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1181). 

cc @GiulioRomualdi @Giulero @gabrielenava @HosameldinMohamed @prashanthr05

traversaro, 2020-04-15:
Clearly then it make sense to have this option enabled for all the "simulated" robots, or at least the one in which the simulated FT sensors do not have any offset to compensate. : )

GiulioRomualdi, 2020-04-15:
That's really cool! I remember when I was young and naive that I spent two hours before understanding that `resetOffset` was required

prashanthr05, 2020-06-11:
I have a question regarding this. I am trying to bypass the initial calibration with simulation by turning a flag on/off from the configuration file.

While doing this, I noted that there is a relevant flag `validOffsetAvailable` that needs to be activated for proper completion of the `publishEstimatedQuantities()` method in order to update the output ports. This flag is turned on by the method endCalibration()
https://github.com/robotology/whole-body-estimators/blob/22302eef59ad2f0b900a662490850206a6aea909/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L2281 which is in turn called by `computeCalibration()` method.
https://github.com/robotology/whole-body-estimators/blob/22302eef59ad2f0b900a662490850206a6aea909/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1884
Does this computeCalibration() method have to be called every iteration ? As done in 
https://github.com/robotology/whole-body-estimators/blob/22302eef59ad2f0b900a662490850206a6aea909/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1884 

or needs to be called only once?

Also. would it be fine to call endCalibration() at the point where I reset the offsets directly?

traversaro, 2020-06-11:
> I have a question regarding this. I am trying to bypass the initial calibration with simulation by turning a flag on/off from the configuration file.
>
> While doing this, I noted that there is a [...]

It needs to be called at avery run, but it does nothing if the calibration is not active:

https://github.com/robotology/whole-body-estimators/blob/22302eef59ad2f0b900a662490850206a6aea909/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1616 .

> 
> Also. would it be fine to call endCalibration() at the point where I reset the offsets directly?

I would simply create a method or set directly `validOffsetAvailable` to `true`, I don't think we need to set also all the other attributes modified by `endCalibration()`, and in general calling `endCalibration` if not calibration was performed would not be clear.

prashanthr05, 2020-06-15:
The initial calibration can be skipped in order to use zero offsets for FT sensors at startup by setting a flag `startWithZeroFTSensorOffsets` to true in the configuration file `devices/wholeBodyDynamics/app/wholebodydynamics-icub-external-six-fts-sim.xml`.

The PR #72 was merged. Closing this issue.

closed 2020-06-15 by prashanthr05