gbionics/whole-body-estimators #73: Link to YARP::YARP_eigen

traversaro, 2020-06-16:
Compiling the baseEstimatorV1 device against a YARP used from the build directory result in the following error:
~~~
[ 45%] Building CXX object devices/baseEstimatorV1/CMakeFiles/baseEstimatorV1.dir/src/baseEstimatorV1.cpp.o
In file included from /usr/local/src/robot/robotology-superbuild/robotology/whole-body-estimators/devices/baseEstimatorV1/src/baseEstimatorV1.cpp:9:0:
/usr/local/src/robot/robotology-superbuild/robotology/whole-body-estimators/devices/baseEstimatorV1/include/baseEstimatorV1.h:32:10: fatal error: yarp/eigen/Eigen.h: No such file or directory
 #include <yarp/eigen/Eigen.h>
~~~
this happens because `baseEstimatorV1` does not correctly links to `YARP::YARP_eigen`.

traversaro, 2020-06-16:
See https://github.com/xEnVrE/robots-modules/blob/38aaa2beeee1e74ca84611d055cc595aacf25a92/src/ICUB/ExtrinsicCalibration/Test/CMakeLists.txt#L32 for an example on how to link `YARP::YARP_eigen`.

prashanthr05, 2020-06-16:
Shouldn't this be available in ${YARP_LIBRARIES}?

https://github.com/robotology/whole-body-estimators/blob/0c4cc016a6860a219f76b19abc51deaa4e2b10f1/devices/baseEstimatorV1/CMakeLists.txt#L39

From what I noted, a YARP installation using robotology-superbuild was creating an `include/yarp/Eigen` folder,  whereas in iCubGenova02 there was no such installation.

traversaro, 2020-06-16:
> Shouldn't this be available in ${YARP_LIBRARIES}?

I am not sure, I guess in any case you need to explicitly ask to it via https://github.com/xEnVrE/robots-modules/blob/38aaa2beeee1e74ca84611d055cc595aacf25a92/src/ICUB/ExtrinsicCalibration/Test/CMakeLists.txt#L16 . 

> whereas in iCubGenova02 there was no such installation.

Using YARP from the build directory is not widespread nowadays, but it still works fine even without any installation.

lrapetti, 2020-06-16:
> #74 has been opened. @lrapetti told he would test this PR on iCubGenova02.
> 
> Thank you @lrapetti.

I have tested the code you commited using YARP from the build directory and it fixes the compilation issue

prashanthr05, 2020-06-16:
Great, thanks a lot @lrapetti !!

prashanthr05, 2020-06-16:
#74 was merged. Closing this issue.

PR #74 (2020-06-16): [devices][baseEstimatorv1] Link to YARP components eigen, os and sig 
Fixes #73.
closed 2020-06-16 by prashanthr05