gbionics/whole-body-estimators #76: Couldn't run baseEstimatorV1

diegoferigo, 2020-06-30:
I'm trying to run the [`baseEstimatorV1`](https://github.com/robotology/whole-body-estimators/tree/38dc6a269f1b0bce09f31d8ae9c4ca3602e8e46b/devices/baseEstimatorV1) device on my setup. 

Following the setup instructions of the README file, I did the following:

1. Start the yarp server
1. Open Gazebo Classic with the clock plugin
1. Insert `iCubGazeboV2_5 (no hands)`
1. Play the simulation
1. Run `YARP_CLOCK=/clock yarprobotinterface --config launch-fbe-analogsens.xml`

<details>
<summary>Click to expand the full output</summary>

```
~ 
❯ YARP_CLOCK=/clock yarprobotinterface --config launch-fbe-analogsens.xml
yarp: Port /IITICUBLAP110/yarprobotinterface/31105/clock:i active at tcp://192.168.8.102:10126/
Success: port-to-port persistent connection added.
yarp: Waiting for clock server to start broadcasting data ...
yarp: Receiving input from /clock to /IITICUBLAP110/yarprobotinterface/31105/clock:i using tcp
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/dferigo/yarprobotinterface.ini] (pwd)
||| checking [/iit/sources/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV2_5/yarprobotinterface.ini] (robot)
||| checking [/iit/sources/robotology-superbuild/build/install/share/iCub/robots/iCubGazeboV2_5/yarprobotinterface.ini] (robot)
||| checking [/iit/sources/robotology-superbuild/build/install/share/ICUBcontrib/robots/iCubGazeboV2_5/yarprobotinterface.ini] (robot)
||| checking [/iit/sources/robotology-superbuild/build/install/share/wbc/robots/iCubGazeboV2_5/yarprobotinterface.ini] (robot)
||| checking [/home/dferigo/.config/yarp/yarprobotinterface.ini] (YARP_CONFIG_HOME)
||| checking [/home/dferigo/.local/share/yarp/yarprobotinterface.ini] (YARP_DATA_HOME)
||| checking [/etc/yarp/yarprobotinterface.ini] (YARP_CONFIG_DIRS)
||| checking [/iit/sources/robotology-superbuild/build/install/share/yarp/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/iit/sources/robotology-superbuild/build/install/share/iCub/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/iit/sources/robotology-superbuild/build/install/share/ICUBcontrib/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/iit/sources/robotology-superbuild/robotology/icub-tests/suites/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/iit/sources/robotology-superbuild/build/install/share/human-gazebo/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/iit/sources/robotology-superbuild/build/install/share/HumanDynamicsEstimation/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/iit/sources/robotology-superbuild/build/install/share/Wearables/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/iit/sources/robotology-superbuild/build/install/share/wbc/yarprobotinterface.ini] (yarp.d)
yarp: cannot read from yarprobotinterface.ini
||| finding file [config]
||| checking [/home/dferigo/launch-fbe-analogsens.xml] (pwd)
||| checking [/iit/sources/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV2_5/launch-fbe-analogsens.xml] (robot)
||| found /iit/sources/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV2_5/launch-fbe-analogsens.xml
[DEBUG]Reading file /iit/sources/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV2_5/launch-fbe-analogsens.xml 
[DEBUG]yarprobotinterface: using xml parser for DTD v3.x 
[DEBUG]Reading file /iit/sources/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV2_5/launch-fbe-analogsens.xml 
[DEBUG]Preprocessor complete in:  0 s 
yarp: Port /baseestimation/yarprobotinterface active at tcp://192.168.8.102:10062/
[INFO]startup phase starting... 
||| finding paths [plugins]
yarp: Port /baseestimation/torso/rpc:o active at tcp://192.168.8.102:10063/
yarp: Port /baseestimation/torso/command:o active at tcp://192.168.8.102:10064/
yarp: Port /baseestimation/torso/stateExt:i active at tcp://192.168.8.102:10065/
yarp: Sending output from /baseestimation/torso/rpc:o to /icubSim/torso/rpc:i using tcp
yarp: Sending output from /baseestimation/torso/command:o to /icubSim/torso/command:i using udp
[INFO]created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
yarp: Receiving input from /icubSim/torso/stateExt:o to /baseestimation/torso/stateExt:i using udp
yarp: Port /baseestimation/left_arm/rpc:o active at tcp://192.168.8.102:10066/
yarp: Port /baseestimation/left_arm/command:o active at tcp://192.168.8.102:10067/
yarp: Port /baseestimation/left_arm/stateExt:i active at tcp://192.168.8.102:10068/
yarp: Sending output from /baseestimation/left_arm/rpc:o to /icubSim/left_arm/rpc:i using tcp
yarp: Sending output from /baseestimation/left_arm/command:o to /icubSim/left_arm/command:i using udp
yarp: Receiving input from /icubSim/left_arm/stateExt:o to /baseestimation/left_arm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
yarp: Port /baseestimation/right_arm/rpc:o active at tcp://192.168.8.102:10069/
yarp: Port /baseestimation/right_arm/command:o active at tcp://192.168.8.102:10070/
yarp: Port /baseestimation/right_arm/stateExt:i active at tcp://192.168.8.102:10071/
yarp: Sending output from /baseestimation/right_arm/rpc:o to /icubSim/right_arm/rpc:i using tcp
yarp: Sending output from /baseestimation/right_arm/command:o to /icubSim/right_arm/command:i using udp
yarp: Receiving input from /icubSim/right_arm/stateExt:o to /baseestimation/right_arm/stateExt:i using udp
[... 210 lines cut ...]
[WARNING]Cannot open device baseestimator 
[WARNING]Cannot open device baseestimator 
[WARNING]There was some problem opening one or more devices. Please check the log and your configuration 
[ERROR]One or more devices failed opening... see previous log messages for more info 
```

</details>

I'm trying to go through the code. It seems that the yarprobotinterface configuration is self-contained and already takes care to open all the depending devices, particularly:

- All the remote control boards and control board remapper (I guess used by wbd)
- The head IMU device (I guess used by the estimator)
- The F/T sensors (I guess used by wbd)

The error is raised by the following code:

https://github.com/robotology/whole-body-estimators/blob/38dc6a269f1b0bce09f31d8ae9c4ca3602e8e46b/devices/baseEstimatorV1/src/fbeRobotInterface.cpp#L432-L438

More details:

- Ubuntu 18.04
- Gazebo 11
- robotology-superbuild updated today

cc @prashanthr05 @traversaro

prashanthr05, 2020-06-30:
Unfortunately, there seems to be a mismatch in that documentation and the configuration file. Apparently, whole-body-dynamics is not launched internally by `launch-fbe-analogsens.xml`.

diegoferigo, 2020-06-30:
It seems to me that `wholeBodyDynamics` does not get started, even if the readme reports:

>   - To run `baseEstimatorV1`,
>
>    ``` sh
>     YARP_CLOCK=/clock yarprobotinterface --config launch-fbe-analogsens.xml
>    ```
>    This launches both the floating base estimation device and the whole body dynamics device.

prashanthr05, 2020-06-30:
Yes! Here we are launching only the baseEstimatorV1 device.  I will open a PR to fix the README.

diegoferigo, 2020-06-30:
Thanks @prashanthr05! Is the reset offset procedure still required? If it is not, maybe you can update also that section.

prashanthr05, 2020-06-30:
Definitely.

prashanthr05, 2020-06-30:
>  Is the reset offset procedure still required?

Sorry, I just managed to skip that question, involuntarily.

FYI, that can be avoided by simply adding a parameter to your WBD config file. 
`startWithZeroFTSensorOffsets` set to true.

which is done [here](https://github.com/robotology/whole-body-estimators/blob/38dc6a269f1b0bce09f31d8ae9c4ca3602e8e46b/devices/wholeBodyDynamics/app/wholebodydynamics-icub-external-six-fts-sim.xml#L17)

diegoferigo, 2020-06-30:
Ok, thanks!

PR #77 (2020-07-01): Update README.md
This PR updates the documentation for launching the `baseEstimatorv1`.

Fixes #76.
closed 2020-07-01 by prashanthr05