gbionics/whole-body-estimators #81: Clarify velocity representation

diegoferigo, 2020-07-06:
The `baseEstimatorV1` devices streams the estimated quantities to the `/base-estimator/floating_base/state:o` port. The methods called are the following, where a minimal documentation exists:

https://github.com/robotology/whole-body-estimators/blob/9608ba9ccb2262ae18244328c4efa1b5aafe5583/devices/baseEstimatorV1/include/baseEstimatorV1.h#L427-L444

It is not clear, however, what is the velocity representation used. Is it the mixed representation as defined in [Multibody Dynamics Notation](https://pure.tue.nl/ws/portalfiles/portal/139293126/A_Multibody_Dynamics_Notation_Revision_2_.pdf)?

cc @prashanthr05 @traversaro

prashanthr05, 2020-07-06:
Yes, internally iDynTree free-floating frame Jacobians are used to compute the velocity assuming rigid contacts at the feet. And these Jacobians are computed using mixed-velocity representations.

diegoferigo, 2020-07-06:
Thanks @prashanthr05 for the confirmation!

PR #82 (2020-07-07): Remove unused code, fix configuration and update documentation 
- Removed a method updateBaseVelocityWithIMU() which was unused and the internal computations seemed to be wrong. (missing an adjoint velocity transform from IMU frame to base frame)

- Removed redundant opening of `controlboard` devices, which is unused by the estimator device, since we open a remotecontrolboardremapper and the estimator device attaches to this remapper. Thanks to @diegoferigo for spotting this.

- Update documentation. Fixes #81.
closed 2020-07-07 by HosameldinMohamed