gbionics/whole-body-estimators #85: Handle CMake deprecation warnings

HosameldinMohamed, 2020-10-07:
When I configure in cmake I get the following warnings (latest commit a02678cca060aa524d14ca8bd0838c1ae5521bac in `devel`):

```
CMake Warning (dev) at libraries/ctrlLibRT/CMakeLists.txt:9 (project):
   Policy CMP0048 is not set: project() command manages VERSION variables.
   Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
   command to set the policy and suppress this warning.

   The following variable(s) would be set to empty:

     PROJECT_VERSION
     PROJECT_VERSION_MAJOR
     PROJECT_VERSION_MINOR
 This warning is for project developers.  Use -Wno-dev to suppress it.

 CMake Deprecation Warning at /home/hosameldin/Github/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:226 (message):
   The YARP_INCLUDE_DIRS variable is deprecated
 Call Stack (most recent call first):
   libraries/ctrlLibRT/CMakeLists.txt:9999 (_yarp_deprecated_variable_warning)
   libraries/ctrlLibRT/CMakeLists.txt:23 (target_include_directories)

 CMake Warning (dev) at idl/wholeBodyDynamicsSettings/CMakeLists.txt:7 (project):
   Policy CMP0048 is not set: project() command manages VERSION variables.
   Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
   command to set the policy and suppress this warning.

   The following variable(s) would be set to empty:

     PROJECT_VERSION
     PROJECT_VERSION_MAJOR
     PROJECT_VERSION_MINOR
 This warning is for project developers.  Use -Wno-dev to suppress it.

 CMake Deprecation Warning at /home/hosameldin/Github/robotology-superbuild/build/install/share/yarp/cmake/YarpIDL.cmake:213 (message):
   Passing variables without name argument to YARP_IDL_TO_DIR is deprecated.
   Use SOURCES_VAR, HEADERS_VAR and INCLUDE_DIRS_VAR
 Call Stack (most recent call first):
   idl/wholeBodyDynamicsSettings/CMakeLists.txt:9 (yarp_idl_to_dir)

 CMake Deprecation Warning at /home/hosameldin/Github/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:226 (message):
   The YARP_INCLUDE_DIRS variable is deprecated
 Call Stack (most recent call first):
   idl/wholeBodyDynamicsSettings/CMakeLists.txt:9999 (_yarp_deprecated_variable_warning)
   idl/wholeBodyDynamicsSettings/CMakeLists.txt:13 (target_include_directories)

 CMake Warning (dev) at idl/wholeBodyDynamics_IDLServer/CMakeLists.txt:7 (project):
   Policy CMP0048 is not set: project() command manages VERSION variables.
   Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
   command to set the policy and suppress this warning.

   The following variable(s) would be set to empty:

     PROJECT_VERSION
     PROJECT_VERSION_MAJOR
     PROJECT_VERSION_MINOR
 This warning is for project developers.  Use -Wno-dev to suppress it.

 CMake Deprecation Warning at /home/hosameldin/Github/robotology-superbuild/build/install/share/yarp/cmake/YarpIDL.cmake:213 (message):
   Passing variables without name argument to YARP_IDL_TO_DIR is deprecated.
   Use SOURCES_VAR, HEADERS_VAR and INCLUDE_DIRS_VAR
 Call Stack (most recent call first):
   idl/wholeBodyDynamics_IDLServer/CMakeLists.txt:9 (yarp_idl_to_dir)

 CMake Deprecation Warning at /home/hosameldin/Github/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:226 (message):
   The YARP_INCLUDE_DIRS variable is deprecated
Call Stack (most recent call first):
   idl/wholeBodyDynamics_IDLServer/CMakeLists.txt:9999 (_yarp_deprecated_variable_warning)
   idl/wholeBodyDynamics_IDLServer/CMakeLists.txt:13 (target_include_directories)

 CMake Deprecation Warning at /home/hosameldin/Github/robotology-superbuild/build/install/share/yarp/cmake/YarpIDL.cmake:213 (message):
   Passing variables without name argument to YARP_IDL_TO_DIR is deprecated.
   Use SOURCES_VAR, HEADERS_VAR and INCLUDE_DIRS_VAR
 Call Stack (most recent call first):
   idl/floatingBaseEstimatorRPC/CMakeLists.txt:5 (yarp_idl_to_dir)

 CMake Deprecation Warning at /home/hosameldin/Github/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:226 (message):
   The YARP_INCLUDE_DIRS variable is deprecated
 Call Stack (most recent call first):
   idl/floatingBaseEstimatorRPC/CMakeLists.txt:9999 (_yarp_deprecated_variable_warning)
   idl/floatingBaseEstimatorRPC/CMakeLists.txt:9 (target_include_directories)

 CMake Deprecation Warning at /home/hosameldin/Github/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:283 (message):
   The YARP_MODULE_PATH variable is deprecated.  CMake find package modules
   are now in YCM.
 Call Stack (most recent call first):
   devices/wholeBodyDynamics/CMakeLists.txt:9999 (_yarp_module_path_is_deprecated)
   devices/wholeBodyDynamics/CMakeLists.txt:10 (list)

 CMake Deprecation Warning at /home/hosameldin/Github/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:226 (message):
   The YARP_INCLUDE_DIRS variable is deprecated
 Call Stack (most recent call first):
   devices/wholeBodyDynamics/CMakeLists.txt:9999 (_yarp_deprecated_variable_warning)
   devices/wholeBodyDynamics/CMakeLists.txt:22 (include_directories)

 CMake Deprecation Warning at /home/hosameldin/Github/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:226 (message):
   The YARP_INCLUDE_DIRS variable is deprecated
 Call Stack (most recent call first):
   devices/virtualAnalogClient/CMakeLists.txt:9999 (_yarp_deprecated_variable_warning)
   devices/virtualAnalogClient/CMakeLists.txt:16 (include_directories)

 CMake Deprecation Warning at /home/hosameldin/Github/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:226 (message):
   The YARP_INCLUDE_DIRS variable is deprecated
 Call Stack (most recent call first):
   devices/virtualAnalogRemapper/CMakeLists.txt:9999 (_yarp_deprecated_variable_warning)
   devices/virtualAnalogRemapper/CMakeLists.txt:16 (include_directories)

 CMake Deprecation Warning at /home/hosameldin/Github/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:226 (message):
   The YARP_INCLUDE_DIRS variable is deprecated
 Call Stack (most recent call first):
   devices/genericSensorClient/CMakeLists.txt:9999 (_yarp_deprecated_variable_warning)
   devices/genericSensorClient/CMakeLists.txt:16 (include_directories)

```

HosameldinMohamed, 2020-10-07:
Comment from @prashanthr05 https://github.com/robotology/whole-body-estimators/pull/58#issuecomment-608297683:

> If we look at the Release notes of YARP,
> 
> The Mutex related classes are deprecated with YARP 3.3 (addressed)
> There is a deprecation of Vocab in YARP 3.1 which was also showing up during compilation. But this was related to autogenerated idl files. I didnt know how to handle this.
> Additionally some CMake deprecations were introduced in YARP 3.0 itself. (unaddressed)

traversaro, 2020-10-09:
Some comments:
* Use of `YARP_INCLUDE_DIRS` can be dropped, as now that is handled by CMake's imported targets
* Use of `YARP_MODULE_PATH` can be dropped, as the path is automatically updated when you do `find_package(YARP)` 
* If you use the idl component of YARP, remember to search for it in  `find_package(YARP)` 
* The `project` call in `ctrlLibRT` can be removed.

prashanthr05, 2020-10-10:
- In libraries `wholeBodyDynamicsHelpers` seem to be empty. They can also be removed.

HosameldinMohamed, 2020-10-13:
> * Use of `YARP_INCLUDE_DIRS` can be dropped, as now that is handled by CMake's imported targets
> * Use of `YARP_MODULE_PATH` can be dropped, as the path is automatically updated when you do [...]

Thanks. I opened PR #87.

> * In libraries `wholeBodyDynamicsHelpers` seem to be empty. They can also be removed.

@prashanthr05 Maybe I can do that in a separate PR? to keep #87 clean?

prashanthr05, 2020-10-14:
> @prashanthr05 Maybe I can do that in a separate PR? to keep #87 clean?

Thanks @HosameldinMohamed! Yes, this is indeed unrelated, even to this issue. My bad.

prashanthr05, 2020-11-03:
Shall we close this issue?

HosameldinMohamed, 2020-11-03:
I think we can since the main goal of the issue is done.

PR #87 (2020-10-13): Handle CMake deprecation warnings
Fixes #85.

I don't think it's necessary to update the CHANGELOG file, what do you think?
closed 2020-11-03 by HosameldinMohamed