gbionics/whole-body-estimators #91: whole-body-dynamics crashes on iCubGazeboV3

GiulioRomualdi, 2020-11-16:
If I try to run `wholebodydynamics` on `iCubGazeboV3` I get the following error. (I'm using the model defined in https://github.com/dic-iit/icub-models/tree/icub3/wbd)

```
gromualdi@iiticublap109:~$ yarprobotinterface --config launch-wholebodydynamics.xml
yarp: Port /iiticublap109/yarprobotinterface/4079/clock:i active at tcp://192.168.1.10:10103/
Success: port-to-port persistent connection added.
yarp: Waiting for clock server to start broadcasting data ...
yarp: Receiving input from /clock to /iiticublap109/yarprobotinterface/4079/clock:i using tcp
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/gromualdi/yarprobotinterface.ini] (pwd)
||| checking [/home/gromualdi/robot-code/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV3/yarprobotinterface.ini] (robot)
||| checking [/home/gromualdi/robot-code/robotology-superbuild/build/install/share/iCub/robots/iCubGazeboV3/yarprobotinterface.ini] (robot)
||| checking [/home/gromualdi/robot-code/robotology-superbuild/build/install/share/ICUBcontrib/robots/iCubGazeboV3/yarprobotinterface.ini] (robot)
||| checking [/home/gromualdi/.config/yarp/yarprobotinterface.ini] (YARP_CONFIG_HOME)
||| checking [/home/gromualdi/.local/share/yarp/yarprobotinterface.ini] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/yarprobotinterface.ini] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/yarprobotinterface.ini] (YARP_CONFIG_DIRS)
||| checking [/home/gromualdi/robot-code/robotology-superbuild/build/install/share/yarp/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/home/gromualdi/robot-code/robotology-superbuild/build/install/share/iCub/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/home/gromualdi/robot-code/robotology-superbuild/build/install/share/ICUBcontrib/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/home/gromualdi/robot-install/share/yarp/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/home/gromualdi/robot-install/share/iCub/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/home/gromualdi/robot-install/share/ICUBcontrib/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/home/gromualdi/robot-code/component_ironcub/models/iRonCub-Mk0/ironBot/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/home/gromualdi/robot-code/component_ironcub/models/iRonCub-Mk1/iRonCub/yarprobotinterface.ini] (YARP_DATA_DIRS)
yarp: cannot read from yarprobotinterface.ini
||| finding file [config]
||| checking [/home/gromualdi/launch-wholebodydynamics.xml] (pwd)
||| checking [/home/gromualdi/robot-code/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV3/launch-wholebodydynamics.xml] (robot)
||| found /home/gromualdi/robot-code/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV3/launch-wholebodydynamics.xml
[DEBUG]Reading file /home/gromualdi/robot-code/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV3/launch-wholebodydynamics.xml 
[WARNING]Invalid syntax while loading /home/gromualdi/robot-code/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV3/launch-wholebodydynamics.xml at line 1 . Expected document of type robot . Found devices 
[DEBUG]yarprobotinterface: using xml parser for DTD v3.x 
[DEBUG]Reading file /home/gromualdi/robot-code/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV3/launch-wholebodydynamics.xml 
[DEBUG]Preprocessor complete in:  0 s 
[WARNING]Invalid syntax while loading  at line 4 . "robot" element should contain the "portprefix" attribute. Using "name" attribute 
yarp: Port /iCubGazeboV3/yarprobotinterface active at tcp://192.168.1.10:10055/
[INFO]startup phase starting... 
||| finding paths [plugins]
yarp: Port /wholeBodyDynamics/torso/rpc:o active at tcp://192.168.1.10:10056/
yarp: Port /wholeBodyDynamics/torso/command:o active at tcp://192.168.1.10:10057/
yarp: Port /wholeBodyDynamics/torso/stateExt:i active at tcp://192.168.1.10:10058/
yarp: Sending output from /wholeBodyDynamics/torso/rpc:o to /icubSim/torso/rpc:i using tcp
yarp: Sending output from /wholeBodyDynamics/torso/command:o to /icubSim/torso/command:i using udp
yarp: Receiving input from /icubSim/torso/stateExt:o to /wholeBodyDynamics/torso/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
yarp: Port /wholeBodyDynamics/left_arm/rpc:o active at tcp://192.168.1.10:10059/
yarp: Port /wholeBodyDynamics/left_arm/command:o active at tcp://192.168.1.10:10060/
yarp: Port /wholeBodyDynamics/left_arm/stateExt:i active at tcp://192.168.1.10:10061/
yarp: Sending output from /wholeBodyDynamics/left_arm/rpc:o to /icubSim/left_arm/rpc:i using tcp
yarp: Sending output from /wholeBodyDynamics/left_arm/command:o to /icubSim/left_arm/command:i using udp
yarp: Receiving input from /icubSim/left_arm/stateExt:o to /wholeBodyDynamics/left_arm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
yarp: Port /wholeBodyDynamics/right_arm/rpc:o active at tcp://192.168.1.10:10062/
yarp: Port /wholeBodyDynamics/right_arm/command:o active at tcp://192.168.1.10:10063/
yarp: Port /wholeBodyDynamics/right_arm/stateExt:i active at tcp://192.168.1.10:10064/
yarp: Sending output from /wholeBodyDynamics/right_arm/rpc:o to /icubSim/right_arm/rpc:i using tcp
yarp: Sending output from /wholeBodyDynamics/right_arm/command:o to /icubSim/right_arm/command:i using udp
yarp: Receiving input from /icubSim/right_arm/stateExt:o to /wholeBodyDynamics/right_arm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
yarp: Port /wholeBodyDynamics/left_leg/rpc:o active at tcp://192.168.1.10:10065/
yarp: Port /wholeBodyDynamics/left_leg/command:o active at tcp://192.168.1.10:10066/
yarp: Port /wholeBodyDynamics/left_leg/stateExt:i active at tcp://192.168.1.10:10067/
yarp: Sending output from /wholeBodyDynamics/left_leg/rpc:o to /icubSim/left_leg/rpc:i using tcp
yarp: Sending output from /wholeBodyDynamics/left_leg/command:o to /icubSim/left_leg/command:i using udp
yarp: Receiving input from /icubSim/left_leg/stateExt:o to /wholeBodyDynamics/left_leg/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
yarp: Port /wholeBodyDynamics/right_leg/rpc:o active at tcp://192.168.1.10:10068/
yarp: Port /wholeBodyDynamics/right_leg/command:o active at tcp://192.168.1.10:10069/
[... 69 lines cut ...]
[INFO]wholeBodyDynamics : Loading model from  /home/gromualdi/robot-code/robotology-superbuild/build/install/share/iCub/robots/iCubGazeboV3/model.urdf 
yarp: Port /wholeBodyDynamics/rpc active at tcp://192.168.1.10:10091/
yarp: Port /wholeBodyDynamics/settings active at tcp://192.168.1.10:10092/
||| finding paths [plugins]
[INFO]created device <controlboardremapper>. See C++ class ControlBoardRemapper for documentation.
||| finding paths [plugins]
[INFO]created device <virtualAnalogRemapper>. See C++ class yarp::dev::VirtualAnalogRemapper for documentation.
yarp: Port /wholeBodyDynamics/left_arm/endEffectorWrench:o active at tcp://192.168.1.10:10093/
yarp: Port /wholeBodyDynamics/right_arm/endEffectorWrench:o active at tcp://192.168.1.10:10094/
yarp: Port /wholeBodyDynamics/left_foot_front/cartesianEndEffectorWrench:o active at tcp://192.168.1.10:10095/
yarp: Port /wholeBodyDynamics/left_foot_rear/cartesianEndEffectorWrench:o active at tcp://192.168.1.10:10096/
yarp: Port /wholeBodyDynamics/right_foot_front/cartesianEndEffectorWrench:o active at tcp://192.168.1.10:10097/
yarp: Port /wholeBodyDynamics/right_foot_rear/cartesianEndEffectorWrench:o active at tcp://192.168.1.10:10098/
||| finding paths [plugins]
[INFO]created device <multipleanalogsensorsremapper>. See C++ class MultipleAnalogSensorsRemapper for documentation.
[INFO]created device <wholebodydynamics>. See C++ class yarp::dev::WholeBodyDynamicsDevice for documentation.
[INFO]Entering action level 15 of phase startup 
[INFO]Starting attach MAS and analog ft 
[DEBUG]wholeBodyDynamicsDevice :: number of ft sensors found in both ft + mas 8 where analog are  8  and mas are  0 
terminate called after throwing an instance of 'std::length_error'
  what():  vector::_M_default_append
Aborted (core dumped)
```

cc @prashanthr05 @traversaro

prashanthr05, 2020-11-16:
This seems to be a bug related to https://github.com/robotology/whole-body-estimators/pull/45/commits/74ffe79c53e90ba040d24503466ba5d7aedf9ba9 that we had fixed with @HosameldinMohamed  in the past.

Could you tell us if you are in the latest `devel`? In my system, with the latest `devel`, it seems to be working.

GiulioRomualdi, 2020-11-16:
yes latest devel: f75d1042fbfd6dfee568f8a186ef9ac452ae764f

do we need a specific version of yarp?

prashanthr05, 2020-11-16:
> do we need a specific version of yarp?

In my local,
YARP version 3.4.100+20201003.1+gitf87738350

I also updated to the latest `master, YARP version 3.4.100+20201111.5+git08c1f164d its fine.

traversaro, 2020-11-16:
Can you collect a stacktrace when the yarprobotinterface goes in segmentation fault? If you are using the superbuild recompiling everything in Debug should be doable, or at least you can recompile just YARP in RelWithDebInfo .

prashanthr05, 2020-11-16:
The output fault is related to MASRemapper.

It must be noted that prior to `yarp v3.4`, there was a bug in MASRemapper which was fixed in https://github.com/robotology/yarp/pull/2300.

GiulioRomualdi, 2020-11-16:
>Maybe updating YARP will fix this issue. Let's see.

Fixed thank you @prashanthr05

closed 2020-11-16 by GiulioRomualdi