gbionics/whole-body-estimators #92: [wholeBodyDynamics] bug in loading settings and model as the estimator is opened

GiulioRomualdi, 2020-11-23:
@prashanthr05 says,

We are trying to use a fixed frame as the kinematics source for the whole body dynamics estimation algorithm.
While testing this, we identified a bug due to the order in which the methods are called within the `open()` function.

The `openEstimator()` method where the iDYnTree model is actually loaded is called after the `loadConfigSettings()` method which contains a check on the frame index using the iDynTree model object.
Please see,

https://github.com/robotology/whole-body-estimators/blob/085d7a6b05d9de207c34aae7b4f64d6080a0a045/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L841-L858

At this point, the model is not loaded and the conditional check against the valid frame index fails, causing the estimator to exit.

traversaro, 2020-11-23:
This make sense. @prashanthr05 do you know how proceed or do you need my input?

prashanthr05, 2020-11-23:
> This make sense. @prashanthr05 do you know how proceed or do you need my input?

I will proceed.

closed 2020-11-24 by prashanthr05