gbionics/whole-body-estimators #96: wholeBodyDynamics::iCubGenova09: Inclusion of VFT wrappers in the launch-wholebody-dynamics causes yarprobotinterface to crash

prashanthr05, 2020-11-27:
I had understood that having the VFT wrappers is required to visualize the contacts  in iCubGui and to publish the estimated joint torques through ports. 
However, the inclusion of the VFT wrappers in the `yarprobotinterface`, as done in [prashanthr05/robots-configuration/feature-wbd](https://github.com/prashanthr05/robots-configuration/blob/d196357f78acf241638c10840f332d3e07f8244e/iCubGenova09/icub_wbd.xml#L72) causes the interface to crash everytime during startup.

@traversaro wrote in https://github.com/dic-iit/component_ANA-Avatar-XPRIZE/issues/68#issuecomment-734824204,

> Virtual Torque wrappers are only useful when using an external WBD, so for now we can just remove them. However, the fact that they make the yarprobointerface publisher could be an indication of some other bug

This needs to be investigated.

cc @GiulioRomualdi  @S-Dafarra

S-Dafarra, 2021-06-30:
Is there any update to this issue?

prashanthr05, 2021-07-02:
> Is there any update to this issue?

No updates on this issue. This got pushed down since we never had use for these wrappers so far. 
Maybe we can close this issue and reopen when necessary.

S-Dafarra, 2021-07-02:
> > Is there any update to this issue?
> 
> No updates on this issue. This got pushed down since we never had use for these wrappers so far.
> Maybe we can close this issue and reopen when necessary.

Since it is most probably a bug, I would keep this open, but remove it from the sprints and the releases.