gbionics/whole-body-estimators #98: Unable to run wholebodydynamics with iCubGazeboV3

GiulioRomualdi, 2020-12-10:
I'm trying to run wbd with the latest model of iCubGazeboV3 https://github.com/robotology/icub-models/commit/93f735875d10053d731b77d5b61df7606380597c
however, I got the following error:
```
yarprobotinterface --config launch-wholebodydynamics.xml
[ERROR] |yarp.os.Property| cannot read from yarprobotinterface.ini
[DEBUG] Reading file /home/gromualdi/robot-code/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV3/launch-wholebodydynamics.xml
[WARNING] Invalid syntax while loading /home/gromualdi/robot-code/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV3/launch-wholebodydynamics.xml at line 1 . Expected document of type robot . Found devices
[DEBUG] yarprobotinterface: using xml parser for DTD v3.x
[DEBUG] Reading file /home/gromualdi/robot-code/robotology-superbuild/build/install/share/yarp/robots/iCubGazeboV3/launch-wholebodydynamics.xml
[DEBUG] Preprocessor complete in:  0.000210285 s
[WARNING] Invalid syntax while loading  at line 4 . "robot" element should contain the "portprefix" attribute. Using "name" attribute
[INFO] |yarp.os.Port| Port /iCubGazeboV3/yarprobotinterface active at tcp://10.255.111.51:10051/
[INFO] startup phase starting...
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/torso/rpc:o active at tcp://10.255.111.51:10052/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/torso/command:o active at tcp://10.255.111.51:10053/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/torso/stateExt:i active at tcp://10.255.111.51:10054/
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/torso/rpc:o to /icubSim/torso/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/torso/command:o to /icubSim/torso/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icubSim/torso/stateExt:o to /wholeBodyDynamics/torso/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/left_arm/rpc:o active at tcp://10.255.111.51:10055/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/left_arm/command:o active at tcp://10.255.111.51:10056/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/left_arm/stateExt:i active at tcp://10.255.111.51:10057/
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/left_arm/rpc:o to /icubSim/left_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/left_arm/command:o to /icubSim/left_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icubSim/left_arm/stateExt:o to /wholeBodyDynamics/left_arm/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/right_arm/rpc:o active at tcp://10.255.111.51:10058/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/right_arm/command:o active at tcp://10.255.111.51:10059/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/right_arm/stateExt:i active at tcp://10.255.111.51:10060/
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/right_arm/rpc:o to /icubSim/right_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/right_arm/command:o to /icubSim/right_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icubSim/right_arm/stateExt:o to /wholeBodyDynamics/right_arm/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/left_leg/rpc:o active at tcp://10.255.111.51:10061/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/left_leg/command:o active at tcp://10.255.111.51:10062/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/left_leg/stateExt:i active at tcp://10.255.111.51:10063/
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/left_leg/rpc:o to /icubSim/left_leg/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/left_leg/command:o to /icubSim/left_leg/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icubSim/left_leg/stateExt:o to /wholeBodyDynamics/left_leg/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/right_leg/rpc:o active at tcp://10.255.111.51:10064/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/right_leg/command:o active at tcp://10.255.111.51:10065/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/right_leg/stateExt:i active at tcp://10.255.111.51:10066/
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/right_leg/rpc:o to /icubSim/right_leg/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/right_leg/command:o to /icubSim/right_leg/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icubSim/right_leg/stateExt:o to /wholeBodyDynamics/right_leg/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/head/rpc:o active at tcp://10.255.111.51:10067/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/head/command:o active at tcp://10.255.111.51:10068/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/head/stateExt:i active at tcp://10.255.111.51:10069/
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/head/rpc:o to /icubSim/head/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/head/command:o to /icubSim/head/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icubSim/head/stateExt:o to /wholeBodyDynamics/head/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/imu active at tcp://10.255.111.51:10070/
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icubSim/inertial to /wholeBodyDynamics/imu using udp
[INFO] |yarp.dev.PolyDriver| Created device <genericSensorClient>. See C++ class yarp::dev::GenericSensorClient for documentation.
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/l_arm_ft_sensor active at tcp://10.255.111.51:10071/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/l_arm_ft_sensor/rpc:o active at tcp://10.255.111.51:10072/
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icubSim/left_arm/analog:o to /wholeBodyDynamics/l_arm_ft_sensor using udp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/l_arm_ft_sensor/rpc:o to /icubSim/left_arm/analog:o/rpc:i using tcp
[INFO] |yarp.dev.PolyDriver| Created device <analogsensorclient>. See C++ class AnalogSensorClient for documentation.
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/r_arm_ft_sensor active at tcp://10.255.111.51:10073/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/r_arm_ft_sensor/rpc:o active at tcp://10.255.111.51:10074/
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icubSim/right_arm/analog:o to /wholeBodyDynamics/r_arm_ft_sensor using udp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /wholeBodyDynamics/r_arm_ft_sensor/rpc:o to /icubSim/right_arm/analog:o/rpc:i using tcp
[INFO] |yarp.dev.PolyDriver| Created device <analogsensorclient>. See C++ class AnalogSensorClient for documentation.
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/l_leg_ft_sensor active at tcp://10.255.111.51:10075/
[INFO] |yarp.os.Port| Port /wholeBodyDynamics/l_leg_ft_sensor/rpc:o active at tcp://10.255.111.51:10076/
[... 125 lines cut ...]
[INFO] |yarp.os.impl.PortCoreOutputUnit| Removing output from /wholeBodyDynamics/torso/command:o to /icubSim/torso/command:i
[INFO] |yarp.os.impl.PortCore| Resetting. m_running = false
[INFO] |yarp.os.impl.PortCoreInputUnit| Removing input from /icubSim/torso/stateExt:o to /wholeBodyDynamics/torso/stateExt:i
[INFO] |yarp.os.impl.PortCore| Resetting. m_running = false
[ERROR] Error in startup phase... see previous messages for more info
[WARNING] Interrupt # 1 # received.
[INFO] Interrupt received. Stopping all running threads.
[... cut ...]
[WARNING] There was some problem running actions for shutdown phase . Please check the log and your configuration
[INFO] shutdown phase finished.
[ERROR] Error in shutdown phase... see previous messages for more info
[INFO] |yarp.os.impl.PortCore| Resetting. m_running = false
[INFO] |yarp.os.RFModule| RFModule failed to open.
```

I think this is related to https://github.com/robotology/icub-models-generator/pull/186 

@prashanthr05 @traversaro if you guide me in the resolution of the problem I can open a PR

prashanthr05, 2020-12-10:
It should work if you comment out these lines, https://github.com/robotology/whole-body-estimators/blob/8f6e78117ccc9f1895d592ab7aa66cb1d9700ae4/devices/wholeBodyDynamics/app/wholebodydynamics-icub-external-eight-fts-sim.xml#L51-L52.

GiulioRomualdi, 2020-12-10:
WIth this patch, I was able to run wbd

```patch
diff --git a/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-eight-fts-sim.xml b/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-eight-fts-sim.xml
index b1abeeb..dc27c79 100644
--- a/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-eight-fts-sim.xml
+++ b/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-eight-fts-sim.xml
@@ -50,15 +50,15 @@
         <param name="local"> /wholeBodyDynamics/r_arm_ft_sensor </param>
     </device>
 
-    <device name="left_upper_leg_strain" type="analogsensorclient">
-        <param name="remote"> /icubSim/left_leg/analog:o </param>
-        <param name="local"> /wholeBodyDynamics/l_leg_ft_sensor </param>
-    </device>
-
-    <device name="right_upper_leg_strain" type="analogsensorclient">
-        <param name="remote"> /icubSim/right_leg/analog:o </param>
-        <param name="local"> /wholeBodyDynamics/r_leg_ft_sensor </param>
-    </device>
+    <!-- <device name="left_upper_leg_strain" type="analogsensorclient"> -->
+    <!--     <param name="remote"> /icubSim/left_leg/analog:o </param> -->
+    <!--     <param name="local"> /wholeBodyDynamics/l_leg_ft_sensor </param> -->
+    <!-- </device> -->
+
+    <!-- <device name="right_upper_leg_strain" type="analogsensorclient"> -->
+    <!--     <param name="remote"> /icubSim/right_leg/analog:o </param> -->
+    <!--     <param name="local"> /wholeBodyDynamics/r_leg_ft_sensor </param> -->
+    <!-- </device> -->
 
     <device name="left_lower_leg_front_strain" type="analogsensorclient">
         <param name="remote"> /icubSim/left_foot_front/analog:o </param>
diff --git a/devices/wholeBodyDynamics/app/wholebodydynamics-icub-external-eight-fts-sim.xml b/devices/wholeBodyDynamics/app/wholebodydynamics-icub-external-eight-fts-sim.xml
index d8efb10..925b035 100644
--- a/devices/wholeBodyDynamics/app/wholebodydynamics-icub-external-eight-fts-sim.xml
+++ b/devices/wholeBodyDynamics/app/wholebodydynamics-icub-external-eight-fts-sim.xml
@@ -27,10 +27,10 @@
         <group name="WBD_OUTPUT_EXTERNAL_WRENCH_PORTS">
             <param name="/wholeBodyDynamics/left_arm/endEffectorWrench:o">(l_hand,l_hand)</param>
             <param name="/wholeBodyDynamics/right_arm/endEffectorWrench:o">(r_hand,l_hand)</param>
-            <param name="/wholeBodyDynamics/left_foot_front/cartesianEndEffectorWrench:o">(l_foot_front,l_foot_front,l_foot_front)</param>
-            <param name="/wholeBodyDynamics/left_foot_rear/cartesianEndEffectorWrench:o">(l_foot_rear,l_foot_rear,l_foot_rear)</param>
-            <param name="/wholeBodyDynamics/right_foot_front/cartesianEndEffectorWrench:o">(r_foot_front,r_foot_front,r_foot_front)</param>
-            <param name="/wholeBodyDynamics/right_foot_rear/cartesianEndEffectorWrench:o">(r_foot_rear,r_foot_rear,r_foot_rear)</param>
+            <param name="/wholeBodyDynamics/left_foot_front/cartesianEndEffectorWrench:o">(l_foot_front,l_sole,l_sole)</param>
+            <param name="/wholeBodyDynamics/left_foot_rear/cartesianEndEffectorWrench:o">(l_foot_rear,l_sole,l_sole)</param>
+            <param name="/wholeBodyDynamics/right_foot_front/cartesianEndEffectorWrench:o">(r_foot_front,r_sole,r_sole)</param>
+            <param name="/wholeBodyDynamics/right_foot_rear/cartesianEndEffectorWrench:o">(r_foot_rear,r_sole,r_sole)</param>
 
         </group>
 
@@ -48,8 +48,8 @@
                 <!-- ft -->
                 <elem name="l_arm_ft_sensor">left_upper_arm_strain</elem>
                 <elem name="r_arm_ft_sensor">right_upper_arm_strain</elem>
-                <elem name="l_leg_ft_sensor">left_upper_leg_strain</elem>
-                <elem name="r_leg_ft_sensor">right_upper_leg_strain</elem>
+                <!-- <elem name="l_leg_ft_sensor">left_upper_leg_strain</elem> -->
+                <!-- <elem name="r_leg_ft_sensor">right_upper_leg_strain</elem> -->
                 <elem name="l_foot_front_ft_sensor">left_lower_leg_front_strain</elem>
                 <elem name="l_foot_rear_ft_sensor">left_lower_leg_rear_strain</elem>
                 <elem name="r_foot_front_ft_sensor">right_lower_leg_front_strain</elem>
```

prashanthr05, 2020-12-10:
Maybe we can just rename the configuration files to be specific to iCub3 and open a PR, to handle this in a clean manner and be in sync with the current developments.

traversaro, 2020-12-10:
We can also just keep the config files for wbd of iCubGenova09 in robots-configuration with all the rest of files.

S-Dafarra, 2021-03-19:
Hi @prashanthr05, @GiulioRomualdi, @traversaro, I just had the same issue. Can we properly fix this with a PR?

closed 2021-03-22 by prashanthr05