gbionics/whole-body-estimators #100: Linking failure against iDynTree devel on macOS

traversaro, 2021-01-29:
Today the Unstable branch jobs on macOS of the superbuild failed with this error:
~~~
2021-01-29T13:58:55.5600220Z FAILED: lib/yarp/baseEstimatorV1.so 
2021-01-29T13:58:55.5716500Z : && /Applications/Xcode_12.3.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -O3 -DNDEBUG -isysroot /Applications/Xcode_12.3.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk -mmacosx-version-min=10.15 -bundle -Wl,-headerpad_max_install_names  -o lib/yarp/baseEstimatorV1.so devices/baseEstimatorV1/CMakeFiles/baseEstimatorV1.dir/src/baseEstimatorV1.cpp.o devices/baseEstimatorV1/CMakeFiles/baseEstimatorV1.dir/src/fbeRobotInterface.cpp.o devices/baseEstimatorV1/CMakeFiles/baseEstimatorV1.dir/src/configureEstimator.cpp.o devices/baseEstimatorV1/CMakeFiles/baseEstimatorV1.dir/src/WalkingLogger.cpp.o devices/baseEstimatorV1/CMakeFiles/baseEstimatorV1.dir/src/Utils.cpp.o devices/baseEstimatorV1/CMakeFiles/baseEstimatorV1.dir/yarp_plugin_baseEstimatorV1.cpp.o  -Wl,-rpath,/Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libYARP_sig.3.4.101.dylib  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libidyntree-solid-shapes.dylib  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libidyntree-inverse-kinematics.dylib  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libidyntree-optimalcontrol.dylib  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libidyntree-visualization.dylib  devices/baseEstimatorV1/libfloatingBaseEstimationRPC-service.a  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libidyntree-high-level.dylib  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libidyntree-estimation.dylib  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libidyntree-modelio-urdf.dylib  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libidyntree-sensors.dylib  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libidyntree-modelio-xml.dylib  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libidyntree-model.dylib  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libidyntree-core.dylib  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libYARP_init.3.4.101.dylib  /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/libYARP_os.3.4.101.dylib && :
2021-01-29T13:58:55.5802290Z Undefined symbols for architecture x86_64:
2021-01-29T13:58:55.5902570Z   "iCub::ctrl::FirstOrderLowPassFilter::setSampleTime(double)", referenced from:
2021-01-29T13:58:55.6003000Z       yarp::dev::baseEstimatorV1::readEncoders(bool) in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.6103480Z   "iCub::ctrl::FirstOrderLowPassFilter::setCutFrequency(double)", referenced from:
2021-01-29T13:58:55.6204260Z       yarp::dev::baseEstimatorV1::setJointVelocityLPFCutoffFrequency(double) in baseEstimatorV1.cpp.o
2021-01-29T13:58:55.6306970Z       non-virtual thunk to yarp::dev::baseEstimatorV1::setJointVelocityLPFCutoffFrequency(double) in baseEstimatorV1.cpp.o
2021-01-29T13:58:55.6349890Z       yarp::dev::baseEstimatorV1::readEncoders(bool) in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.6454340Z   "iCub::ctrl::FirstOrderLowPassFilter::FirstOrderLowPassFilter(double, double, yarp::sig::VectorOf<double> const&)", referenced from:
2021-01-29T13:58:55.6557350Z       yarp::dev::baseEstimatorV1::readEncoders(bool) in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.6659880Z   "yarp::dev::PolyDriver::open(yarp::os::Searchable&)", referenced from:
2021-01-29T13:58:55.6762040Z       yarp::dev::baseEstimatorV1::loadTransformBroadcaster() in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.6864280Z   "yarp::dev::PolyDriver::close()", referenced from:
2021-01-29T13:58:55.6966410Z       yarp::dev::baseEstimatorV1::closeDevice() in baseEstimatorV1.cpp.o
2021-01-29T13:58:55.7068480Z   "yarp::dev::PolyDriver::PolyDriver()", referenced from:
2021-01-29T13:58:55.7170690Z       yarp::dev::baseEstimatorV1::baseEstimatorV1(double, yarp::os::ShouldUseSystemClock) in baseEstimatorV1.cpp.o
2021-01-29T13:58:55.7273250Z       yarp::dev::baseEstimatorV1::baseEstimatorV1() in baseEstimatorV1.cpp.o
2021-01-29T13:58:55.7375370Z   "yarp::dev::PolyDriver::~PolyDriver()", referenced from:
2021-01-29T13:58:55.7509410Z       yarp::dev::baseEstimatorV1::~baseEstimatorV1() in baseEstimatorV1.cpp.o
2021-01-29T13:58:55.7609780Z   "yarp::dev::IMultipleWrapper::~IMultipleWrapper()", referenced from:
2021-01-29T13:58:55.7710290Z       yarp::dev::baseEstimatorV1::baseEstimatorV1(double, yarp::os::ShouldUseSystemClock) in baseEstimatorV1.cpp.o
2021-01-29T13:58:55.7812240Z       yarp::dev::baseEstimatorV1::baseEstimatorV1() in baseEstimatorV1.cpp.o
2021-01-29T13:58:55.7912670Z       yarp::dev::baseEstimatorV1::~baseEstimatorV1() in baseEstimatorV1.cpp.o
2021-01-29T13:58:55.8012780Z   "yarp::dev::PolyDriver::isValid() const", referenced from:
2021-01-29T13:58:55.8113240Z       yarp::dev::baseEstimatorV1::closeDevice() in baseEstimatorV1.cpp.o
2021-01-29T13:58:55.8215110Z   "yarp::dev::PolyDriverList::size() const", referenced from:
2021-01-29T13:58:55.8315780Z       yarp::dev::baseEstimatorV1::attachAll(yarp::dev::PolyDriverList const&) in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.8416440Z       yarp::dev::baseEstimatorV1::attachAllControlBoards(yarp::dev::PolyDriverList const&) in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.8517020Z       yarp::dev::baseEstimatorV1::attachAllInertialMeasurementUnits(yarp::dev::PolyDriverList const&) in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.8617220Z       yarp::dev::baseEstimatorV1::attachAllForceTorqueSensors(yarp::dev::PolyDriverList const&) in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.8718930Z   "yarp::dev::PolyDriverList::operator[](int) const", referenced from:
2021-01-29T13:58:55.8819020Z       yarp::dev::baseEstimatorV1::attachAll(yarp::dev::PolyDriverList const&) in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.8919620Z       yarp::dev::baseEstimatorV1::attachAllControlBoards(yarp::dev::PolyDriverList const&) in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.9021360Z       yarp::dev::baseEstimatorV1::attachAllInertialMeasurementUnits(yarp::dev::PolyDriverList const&) in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.9120880Z       yarp::dev::baseEstimatorV1::attachAllForceTorqueSensors(yarp::dev::PolyDriverList const&) in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.9221310Z   "typeinfo for yarp::dev::IAnalogSensor", referenced from:
2021-01-29T13:58:55.9322840Z       yarp::dev::baseEstimatorV1::attachAllForceTorqueSensors(yarp::dev::PolyDriverList const&) in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.9422630Z   "typeinfo for yarp::dev::IGenericSensor", referenced from:
2021-01-29T13:58:55.9524950Z       yarp::dev::baseEstimatorV1::attachAllInertialMeasurementUnits(yarp::dev::PolyDriverList const&) in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.9624810Z   "typeinfo for yarp::dev::IFrameTransform", referenced from:
2021-01-29T13:58:55.9726190Z       yarp::dev::baseEstimatorV1::loadTransformBroadcaster() in fbeRobotInterface.cpp.o
2021-01-29T13:58:55.9827130Z   "typeinfo for yarp::dev::IMultipleWrapper", referenced from:
2021-01-29T13:58:55.9928860Z       typeinfo for yarp::dev::baseEstimatorV1 in baseEstimatorV1.cpp.o
2021-01-29T13:58:56.0029010Z   "vtable for yarp::dev::IMultipleWrapper", referenced from:
2021-01-29T13:58:56.0130010Z       yarp::dev::baseEstimatorV1::baseEstimatorV1(double, yarp::os::ShouldUseSystemClock) in baseEstimatorV1.cpp.o
2021-01-29T13:58:56.0230930Z       yarp::dev::baseEstimatorV1::baseEstimatorV1() in baseEstimatorV1.cpp.o
2021-01-29T13:58:56.0332460Z   NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
2021-01-29T13:58:56.0431570Z ld: symbol(s) not found for architecture x86_64
2021-01-29T13:58:56.0532770Z clang: error: linker command failed with exit code 1 (use -v to see invocation)
2021-01-29T13:58:56.0634120Z [29/37] Building CXX object devices/wholeBodyDynamics/CMakeFiles/wholeBodyDynamicsDevice.dir/yarp_plugin_wholebodydynamics.cpp.o
[... cut ...]
2021-01-29T13:58:56.0935180Z [32/37] Building CXX object devices/wholeBodyDynamics/CMakeFiles/wholeBodyDynamicsDevice.dir/WholeBodyDynamicsDevice.cpp.o
2021-01-29T13:58:56.1035800Z ninja: build stopped: subcommand failed.
~~~

Probably this is a regression due to https://github.com/robotology/idyntree/pull/807, but that is exposing a problem in this repo as some library/plugin was using the libYARP_dev and ctrlLib functionalities, but without linking it explicitly, so it could make sense to fix the problem here.

traversaro, 2021-01-29:
See https://github.com/robotology/robotology-superbuild/runs/1790760012 for the failure.

traversaro, 2021-01-29:
Note that the problem on macOS arises at linking time, but probably on other problems this missing linking could create problems at runtime.

prashanthr05, 2021-02-01:
Related issue/PR:
- https://github.com/robotology/whole-body-estimators/pull/74
- https://github.com/robotology/whole-body-estimators/issues/73

closed 2021-02-01 by prashanthr05

traversaro, 2021-02-01:
Can you forward merge master in devel? Because the default branch is the one used in the unstable branches of superbuild, by doing that we should be able to fix the Unstable build of the superbuild.

prashanthr05, 2021-02-01:
Yes. I am on it now.

prashanthr05, 2021-02-01:
Done!