gbionics/whole-body-estimators #110: Allow opening and using the contacts:o port without skinDynLib

S-Dafarra, 2021-04-16:
At the moment, the ``contacts:o`` port is opened only if ``useSkinForContacts`` is set to true. See https://github.com/robotology/whole-body-estimators/blob/353985f3997e89dbc6744105077765457e2e0227/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1469-L1482 and https://github.com/robotology/whole-body-estimators/blob/353985f3997e89dbc6744105077765457e2e0227/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L569-L575

I tried to set ``useSkinForContacts`` to ``true`` without setting the ``IDYNTREE_SKINDYNLIB_LINKS``, but then I had a lot of these errors related to https://github.com/robotology/whole-body-estimators/blob/353985f3997e89dbc6744105077765457e2e0227/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L2534 coming from https://github.com/robotology/idyntree/blob/db312bfe9f490f991250603f05cea84a8c156899/src/icub/include/iDynTree/skinDynLibConversionsImplementation.h#L197.

Even if the error seems harmless, this is not sustainable since the ``wholeBodyDynamics`` device runs as a part of the robot main ``yarprobotinterface``, thus filling up the logger in a very short time.

I have tried understanding how to set the ``IDYNTREE_SKINDYNLIB_LINKS`` group, but I could not figure out a correct configuration. In addition, it requires to specify a set of numbers that I no clue how to get right, especially for a robot like iCub3 that is covered only in a small portion by the skin.

Any hint @traversaro @prashanthr05?

prashanthr05, 2021-04-16:
I have never used the skin with WBD so far. But maybe looking into [skinDynLib](https://github.com/robotology/icub-main/tree/master/src/libraries/skinDynLib/include/iCub/skinDynLib) could possibly help?

S-Dafarra, 2021-04-16:
> I have never used the skin with WBD so far. But maybe looking into [skinDynLib](https://github.com/robotology/icub-main/tree/master/src/libraries/skinDynLib/include/iCub/skinDynLib) and related configuration files could possibly help?

Would you have any input on how to get the numbers to put in there, like https://github.com/robotology/robots-configuration/blob/f3fef4277717ae87d825a9a46d994ccd21fef3ec/iCubGenova04/estimators/wholebodydynamics.xml#L24-L40?

S-Dafarra, 2021-04-16:
After a F2F discussion with @traversaro, a possibility would be to open yet another port publishing the content of ``estimateExternalContactWrenches``.

prashanthr05, 2021-04-16:
> Would you have any input on how to get the numbers to put in there, like [...]

On some reading, I think this file might have some relevant information, https://github.com/robotology/icub-main/blob/master/src/libraries/skinDynLib/include/iCub/skinDynLib/common.h#L43 related to the body part enums.

S-Dafarra, 2021-04-16:
> On some reading, I think this file might have some relevant information, https://github.com/robotology/icub-main/blob/master/src/libraries/skinDynLib/include/iCub/skinDynLib/common.h#L43 related to [...]

Ah nice finding!

S-Dafarra, 2021-04-16:
> Ah nice finding!

While the first number is more or less clear, I am not sure what the second number is :thinking:

traversaro, 2021-04-16:
> While the first number is more or less clear, I am not sure what the second number is 🤔

It used to be a reference to the link index in the corresponding iKin Chain, so for `LEFT_ARM` https://github.com/robotology/icub-main/blob/4215771441724b7ece6412d56d7f884f70cca703/src/libraries/iKin/include/iCub/iKin/iKinFwd.h#L1081, `LEFT_LEG` https://github.com/robotology/icub-main/blob/4215771441724b7ece6412d56d7f884f70cca703/src/libraries/iKin/include/iCub/iKin/iKinFwd.h#L1235, etc etc. In reality they refer to the equivalent data structures in iDyn . See https://github.com/robotology/icub-main/blob/master/src/libraries/iDyn/src/iDynContact.cpp#L93 for the code that used it.

traversaro, 2021-04-16:
In https://github.com/robotology/icub-main/pull/462 you can find a PR that we should revive to finally end this confusion.