gbionics/whole-body-estimators #122: wholebodydynamics: Add support to log internal quantities useful for debug

traversaro, 2021-09-09:
As mentioned in https://github.com/robotology/whole-body-estimators/issues/119, it often happens that the estimation provides unexpected values. It would be great to modify `wholebodydynamics` to sistematically log relevant data, to easily debug estimation problems. 

Ideally, the data logged could go in the same file created by the `telemetryDeviceDumper`, to automatically synchronize it with sensor measures. A possible strategy (but possible alternative strategy are welcome) is to: 
* Modify `telemetryDeviceDumper` to expose a C++ YARP device interface that can be used by arbitrary devices to log variables without the need to modify `telemetryDeviceDumper` (`telemetryDeviceDumper`  just need to be manually modified for devices that we don't want to modify, for example because they are hosted in YARP and this would create a circular dependency.
* Once `telemetryDeviceDumper` exposes the necessary interfaces, modify the wbd device to (optinally) log its data via this interface.

prashanthr05, 2021-09-10:
Related:

@isorrentino worked in this direction from BipedalLocomotionFramework side as discussed in https://github.com/ami-iit/element_torque-control-via-current/issues/194#issuecomment-910346899.

A PR is open in BLF: https://github.com/ami-iit/bipedal-locomotion-framework/pull/405

traversaro, 2021-09-10:
> A PR is open in BLF: [ami-iit/bipedal-locomotion-framework#405](https://github.com/ami-iit/bipedal-locomotion-framework/pull/405)

The idea is that by logging directly in WBD, we can log quantities that are not available otherwise (for example input quantities **after the filter**).