gbionics/whole-body-estimators #132: Release master branch?

traversaro, 2021-12-03:
We need to soon release robotology distro 2021.11.0 (as you may notice from the name, we are slightly delayed), could we do a release of the master branch so we package it in 2021.11.0 ? If necessary I can do it myself, thanks! 

@HosameldinMohamed @prashanthr05

Nicogene, 2021-12-03:
For me, it would be nice to have it in order to have #129 in the Distro, but it is not essential.

HosameldinMohamed, 2021-12-03:
> We need to soon release robotology distro 2021.11.0 (as you may notice from the name, we are slightly delayed), could we do a release of the master branch so we package it in 2021.11.0 ? If necessary I can do it myself, thanks! 
> 
> @HosameldinMohamed @prashanthr05 

Hi @traversaro that's cool. I can do it in the evening. But feel free to do it if you are in a hurry. Thanks.

traversaro, 2021-12-03:
No problem, if you can do it in the evening it is fine, thanks!

HosameldinMohamed, 2021-12-03:
New release https://github.com/robotology/whole-body-estimators/releases/tag/v0.6.0

closed 2021-12-06 by Nicogene