gbionics/whole-body-estimators #133: Access the joint torques estimated by WBD

isorrentino, 2021-12-03:
Hi all, I was wondering if there is a way to read the joint torques estimated by WBD when the robot is simulated? I checked the available YARP ports and none is used for the estimated joint torques. Are they exposed in some way? I would like to read them in order to log a dataset.

cc @GiulioRomualdi @traversaro

traversaro, 2021-12-03:
The estimated torques are published in C++ by attaching devices that implement the YARP Device interface `IVirtualAnalogSensor` (see https://github.com/robotology/whole-body-estimators/blob/64b513cbad7a868e20775f71516e5c81dc2c1e80/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1501). This interface is implemented on the real robot on devics such as `embObjMotionControl` (see https://github.com/robotology/icub-main/blob/624bc968dabe64bfb129a25f72e0b80fc9503af0/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L3733), but to publish that on a YARP port, you can use a `virtualAnalogClient` device, see https://github.com/robotology/whole-body-estimators/blob/353985f3997e89dbc6744105077765457e2e0227/devices/virtualAnalogClient/VirtualAnalogClient.h#L36 for its parameters.

See for example:
* https://github.com/robotology/whole-body-estimators/blob/353985f3997e89dbc6744105077765457e2e0227/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-six-fts.xml#L58
* https://github.com/robotology/whole-body-estimators/blob/64b513cbad7a868e20775f71516e5c81dc2c1e80/devices/wholeBodyDynamics/app/wholebodydynamics-icub-external-and-can-six-fts.xml#L58

traversaro, 2021-12-03:
fyi @Nicogene