gbionics/whole-body-estimators #156: [iCubGazeboV3] Uncorrect estimated external wrench

GiulioRomualdi, 2022-08-30:
I spawned iCubGazeboV3 in gazebo and I ran wholebody dynamics with `yarprobotinterface --config launch-wholebodydynamics.xml`.
The robot was in double support and the feet were in contact with the ground

However when I tried to read the contact forces with `yarp read ... /wholeBodyDynamics/left_foot_front/cartesianEndEffectorWrench:o` I got the following result.

```
-6.61824911262959858504 22.7513259601726787196 -19.8714123575537797706 -3.94523009229582255486 -0.386628332966471921495 0.835996243860451682117
-6.70601412783409056573 22.8582984108054354522 -19.8699358998094659512 -3.95174154470659821925 -0.390805641636702727482 0.845556781194174433836
-6.79496840578879535144 22.9110749466858649726 -19.8054163161445622166 -3.95648009198770456152 -0.399091541687972561903 0.850312911042904850945
```

I asked @paolo-viceconte to replicate the same experiment and he got the same result
My setup is
- icub-models https://github.com/robotology/icub-models/commit/74cbcda6e87534afa236187db52f2192e9a7e3f4
- whole-body-estimators https://github.com/robotology/whole-body-estimators/commit/63785f1fa63fb90db1a42a2c326572e10eedd89e
- gazebo-yarp-plugins https://github.com/robotology/gazebo-yarp-plugins/commit/48f0da60c1ed20a4afe7f85536304679ab387ab7

cc @traversaro

GiulioRomualdi, 2022-08-30:
I've just noticed that I've the same problem with iCubGazeboV2_5 so I will rename the issue

GiulioRomualdi, 2022-08-30:
Since the version v0.7.0 is working without any problem I played with `git bisect` This is the first breaking commit 9bd947c6477e686b3b0bba435b4ef64a74bfd88a
```
commit 9bd947c6477e686b3b0bba435b4ef64a74bfd88a
Author: Carlotta Sartore <56030908+CarlottaSartore@users.noreply.github.com>
Date:   Fri Jul 29 11:37:19 2022 +0200

    Introduce MAS Read in whole body estimator  (#153)
    
    * wholeBodyDynamicsDevice: Read from MAS FT sensors
    
    * WIP fix segfaults - to check valid measures
    
    * Fix bugs in reading from MAS interface
    
    * Remove stray commented line
    
    * Update CHANGELOG
    
    Co-authored-by: Prashanth <prashanth.ramadoss@iit.it>
    Co-authored-by: iCubGenova09 <icubgenova09.user@example.com>

 CHANGELOG.md                                       |   1 +
 .../wholeBodyDynamics/WholeBodyDynamicsDevice.cpp  | 103 ++++++++++++++++-----
 .../wholeBodyDynamics/WholeBodyDynamicsDevice.h    |   7 ++
```

traversaro, 2022-08-30:
My intuition is that there is something wrong in the ft serialization, we had a similar problem (if I recall correctly) when debugging https://github.com/robotology/whole-body-estimators/pull/153/files in the first place.

traversaro, 2022-08-30:
In the `iCubGazeboV2_5` test did you changed the `YARP_ROBOT_NAME` ?

traversaro, 2022-08-30:
A possible bug is in https://github.com/robotology/whole-body-estimators/blob/63785f1fa63fb90db1a42a2c326572e10eedd89e/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L2058, that should be:
~~~
auto ftID = std::distance(ftAnalogSensorNames.begin(), ftMasITer);
~~~
instead.

traversaro, 2022-08-30:
PR: https://github.com/robotology/whole-body-estimators/pull/157 . @GiulioRomualdi

GiulioRomualdi, 2022-08-31:
Fixed in #157 thank you @traversaro

closed 2022-08-31 by GiulioRomualdi