gbionics/whole-body-estimators #163: Test attaching MAS FTs to `wholeBodyDynamics`

HosameldinMohamed, 2022-10-04:
## Testing with `wholeBodyDynamics`

I tried to test the device with WBD. First I tried to attach FT sensors as MAS devices but I was getting an error.

Here's the config file I used

<details>
<summary>config file</summary>

```xml
<?xml version="1.0" encoding="UTF-8" ?>
<!DOCTYPE devices PUBLIC "-//YARP//DTD yarprobotinterface 3.0//EN" "http://www.yarp.it/DTD/yarprobotinterfaceV3.0.dtd">

<robot name="@WBD_YARP_ROBOT_NAME">
    <!-- controlboards -->
    <device name="torso_mc" type="remote_controlboard">
        <param name="remote"> /icub/torso </param>
        <param name="local"> /wholeBodyDynamics-thrEst/torso </param>
    </device>

    <device name="left_arm_mc" type="remote_controlboard">
        <param name="remote"> /icub/left_arm </param>
        <param name="local"> /wholeBodyDynamics-thrEst/left_arm </param>
    </device>

    <device name="right_arm_mc" type="remote_controlboard">
        <param name="remote"> /icub/right_arm </param>
        <param name="local"> /wholeBodyDynamics-thrEst/right_arm </param>
    </device>

    <device name="left_leg_mc" type="remote_controlboard">
        <param name="remote"> /icub/left_leg </param>
        <param name="local"> /wholeBodyDynamics-thrEst/left_leg </param>
    </device>

    <device name="right_leg_mc" type="remote_controlboard">
        <param name="remote"> /icub/right_leg </param>
        <param name="local"> /wholeBodyDynamics-thrEst/right_leg </param>
    </device>

    <device name="head_mc" type="remote_controlboard">
        <param name="remote"> /icub/head </param>
        <param name="local"> /wholeBodyDynamics-thrEst/head </param>
    </device>

    <!-- imu -->
    <device name="inertial" type="genericSensorClient">
        <param name="remote"> /icub/inertial </param>
        <param name="local"> /wholeBodyDynamics-thrEst/imu </param>
    </device>

    <!-- FTs -->
     <device xmlns:xi="http://www.w3.org/2001/XInclude" name="left-arm-ft-client" type="multipleanalogsensorsclient">
        <param name="remote">/icub/left_arm</param>
        <param name="local">/wholeBodyDynamics-thrEst/l_arm_ft_sensor</param>
    </device>

    <device xmlns:xi="http://www.w3.org/2001/XInclude" name="right-arm-ft-client" type="multipleanalogsensorsclient">
        <param name="remote">/icub/right_arm</param>
        <param name="local">/wholeBodyDynamics-thrEst/r_arm_ft_sensor</param>
    </device>

    <device name="left_upper_leg_strain" type="analogsensorclient">
        <param name="remote"> /icub/left_leg/analog:o </param>
        <param name="local"> /wholeBodyDynamics-thrEst/l_leg_ft_sensor </param>
    </device>

    <device name="right_upper_leg_strain" type="analogsensorclient">
        <param name="remote"> /icub/right_leg/analog:o </param>
        <param name="local"> /wholeBodyDynamics-thrEst/r_leg_ft_sensor </param>
[... 121 lines cut ...]
</details>

Basically, I replaced 2 `analogServer` FTs with `multipleanalogservers`.

I am getting the following error 

```
[ERROR] WholeBodyDynamicsDevice::attachAll Invalid number of MAS FT sensors after remapper
```

It happens due to this line:

https://github.com/robotology/whole-body-estimators/blob/c46c1852974776038a75b7c0e542ccb2917d7c7f/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1743-L1747

Which is strange because the MAS sensors are detected based on 

```
[DEBUG] wholeBodyDynamicsDevice :: number of ft sensors found in both ft + mas 4 where analog are  2  and mas are  2
```

From line https://github.com/robotology/whole-body-estimators/blob/c46c1852974776038a75b7c0e542ccb2917d7c7f/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1720

Which means `ftSensorList.size()` is actually not empty.

What could I be possibly missing?

HosameldinMohamed, 2022-10-04:
CC @traversaro

traversaro, 2022-10-04:
If I recall correctly the current logic is that you should either have all sensors exposed via IAnalogSensors or all via ISixAxisForceTorqueSensors, you can't mix some sensors exposed via IAnalogSensors  and some others via ISixAxisForceTorqueSensors.

HosameldinMohamed, 2022-10-05:
> If I recall correctly the current logic is that you should either have all sensors exposed via IAnalogSensors or all via ISixAxisForceTorqueSensors, you can't mix some sensors exposed via IAnalogSensors and some others via ISixAxisForceTorqueSensors.

I'm not sure this is the cause of this particular error, anyhow I tried with all FTs exposed via `ISixAxisForceTorqueSensors`, but still the same [error](https://github.com/robotology/whole-body-estimators/issues/163#issue-1396820909) occurs.

The error happens because attach methods in https://github.com/robotology/whole-body-estimators/blob/c46c1852974776038a75b7c0e542ccb2917d7c7f/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1734-L1735

are executed with no errors but still return 0 devices because `remappedMASInterfaces.ftMultiSensors->getNrOfSixAxisForceTorqueSensors()` returns 0.

I found that I am missing the group `multipleAnalogSensorsNames` in the config files, which I guess should list the names of the MAS FT sensors, see this logic
https://github.com/robotology/whole-body-estimators/blob/c46c1852974776038a75b7c0e542ccb2917d7c7f/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L772-L795

But I still can't get it right because I don't know how to properly add this group.
CC @CarlottaSartore @GiulioRomualdi

HosameldinMohamed, 2022-10-05:
> But I still can't get it right because I don't know how to properly add this group.

This configuration worked (we are using 4 FTs in iRonCub):

```xml
<group name="multipleAnalogSensorsNames">
            <param name="SixAxisForceTorqueSensorsNames">("l_arm_ft_sensor", "r_arm_ft_sensor", "l_leg_ft_sensor", "r_leg_ft_sensor")</param>
            <param name="TemperatureSensorsNames">("l_arm_ft_sensor", "r_arm_ft_sensor", "l_leg_ft_sensor", "r_leg_ft_sensor")</param>
</group>
```

Probably we need to document this somewhere..

I was also able to launch the device successfully with mixing attaching `multipleanalogservers` and `analogservers`.

CC @traversaro

HosameldinMohamed, 2023-07-27:
I think we can close this issue!

closed 2023-07-27 by HosameldinMohamed