gbionics/whole-body-estimators #170: [WholeBodyDynamics] change logic of attaching FTs when both AnalogServer and MultipleAnalogServer interfaces are implemented

HosameldinMohamed, 2023-07-27:
When using `WholeBodyDynamics` to attach the gazebo plugin `gazebo_yarp_forcetorque` directly, without `multipleanalogsernsorsserver`, it considers the sensor as both AnalogServer and MultipleAnalogSensorServer device, and attaches it twice.

Maybe we could change the logic and do like what we did with the IMU (using a CMake option `HW_USE_MAS_IMU`)?

HosameldinMohamed, 2023-07-27:
CC @traversaro @GiulioRomualdi @CarlottaSartore

GiulioRomualdi, 2023-07-27:
Can we completely drop analog?

HosameldinMohamed, 2023-07-27:
> Can we completely drop analog?

Would be cleaner for sure!

traversaro, 2023-07-31:
> Can we completely drop analog?

I totally agree, after https://github.com/robotology/icub-models/issues/198 we should be ready to do that, at least for FT sensors.

HosameldinMohamed, 2024-01-02:
`IAnalogSensor` is removed from the devices used for iCub FTs

https://github.com/robotology/icub-main/pull/928

HosameldinMohamed, 2024-03-06:
> Can we completely drop analog?

@traversaro is handling it in #179

traversaro, 2024-03-06:
> > Can we completely drop analog?
> 
> @traversaro is handling it in #179

Good point, I guess that PR also closes this issue, right?

closed 2024-03-06 by traversaro